EdgeTX/edgetx #3468: (cpn) LUA no debug printout possible

ParkerEde, 2023-04-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

with the current main it is no longer possible to print debug messages in the simulator's debug window from LUA Widgets or Scripts

### Expected Behavior

print ("test") should print "test" in simulator debug window from LUA scripts or LUA widgets

### Steps To Reproduce

1. start simulator
2. load a lua widget with print command or start lua script with print command
3. debug window shows no print out

### Version

Nightly (Please give date/commit below)

### Transmitter

Other (Please specify below)

### Operating System (OS)

Windows

### OS Version

windows 11 pro 22H2

### Anything else?

current main

mha1, 2023-04-12:
I can confirm the problem on W10-64Bit and started investigating. It seems fwrite() to stdout is no longer working. Using TRACE_DEBUG_WP() LUA print works:

radio\src\thirdparty\Lua\src\luaconf.h change lines 224-234 to this will bring back LUA print to Simulator Debug console:
```c//#if defined(LUA_LIB) || defined(lua_c)
//#include <stdio.h>
//#define luai_writestring(s,l)	fwrite((s), sizeof(char), (l), stdout)
//#define luai_writeline()	(luai_writestring("\n", 1), fflush(stdout))
//#define luai_writestringerror(s,p) \
//        (fprintf(stderr, (s), (p)), fflush(stderr))
//#else
#define luai_writestring(s,l)       TRACE_DEBUG_WP("%s", s);
#define luai_writeline()            TRACE_DEBUG_WP("\n");
#define luai_writestringerror(s,p)  TRACE_DEBUG_WP(s, p);
//#endif
```
Please let me know if this might be a valid fix but I'd rather understand why fwrite() to stdout stopped working. Any ideas?

mha1, 2023-04-12:
Actually https://github.com/EdgeTX/edgetx/pull/2994 added `#define LUA_LIB`'s to some `radio/src/api_*.cpp` files which forces the section of code above in `radio\src\thirdparty\Lua\src\luaconf.h` into the #if branch and with this to use fwrite(). But it seems the defines are necessary to  set up the api_*.cpp lib tables. Removing the defines in api_*.cpp is not an option.

The above code is a fix but is it ok to change this thirdparty file?

pfeerick, 2023-04-13:
It seems to be ok to make changes there, there have been other changes and it's not submodule code.

closed 2023-04-17 by pfeerick