EdgeTX/edgetx #3469: compagnon 2.8.2 simulator crash

LS-67, 2023-04-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When choosing "simulation" for a model in compagnon 2.8.2, ...compagnon crash

### Expected Behavior

Compagnon doesn't crash when choosing simulation

### Steps To Reproduce

1) Open Compagnon 2.8.2
2) Select a model.
3) Double click on it
4) Choose "simulation"
and ...compagnon crash 

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

10

FearlessFlyer, 2023-04-12:
Same here on Windows 7 Pro.
I have gone back to the previous version, wich runs well.

mha1, 2023-04-12:
Does the active model have MPM (Multi) selected as internal module? If yes this is fixed with https://github.com/EdgeTX/edgetx/pull/3426 and will be part of 2.8.3.

LS-67, 2023-04-13:
Yes, the model has MPM selected. Thank's

pfeerick, 2023-04-13:
Duplicate of #3469

closed 2023-04-13 by pfeerick