EdgeTX/edgetx #3474: Images folder organized in sub-folders

hdlbq, 2023-04-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Describe the solution you'd like

It will be easier to manage many pictures, for various kind of models (drones, gliders,…) and for the system (splash screens) if they can be sorted in sub-folders.

### Describe alternatives you've considered

I organize currently my pictures by prefixing their names with the initial of their kind. But it is limited by the maximal length of picture file names.

ggoraa, 2023-04-13:
An even better solution is to just bring up the file browser when selecting an image

hdlbq, 2023-04-14:
Thanks for your answer. I'm a newbie in EdgeTx. What should I do to bring up a file brower ?

ggoraa, 2023-04-14:
No, sorry for confusion 😅
I meant that to fix this issue it would be better to make that button bring up the file browser instead of just a list of files in the images folder

pfeerick, 2023-04-14:
Well, not really... then you might go putting the images *anywhere*. I think the real initial problem here was that it seems the file list was allowing you to see invalid files, which I thought were supposed to be filtered out (I bet it is, and the character limit is wrong - I have 16 character limit in my head for some reason, when it seems it should be 14). But I agree it should be possible to then view subfolders. I guess that would actually require there *be* a file browser that could be used here, and actually in a few other places. 🤭