EdgeTX/edgetx #3477: Tx16S Use Roller to Setup FRM303, UI components not follow the order of display

richardclli, 2023-04-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

External RF Setuo, when setting up the FRM303, hard to address the type using the roller:

Mode: FLYSKY -> AFHDS3 ->Channel Range CH1 (skipped the type fields) ->CH18 ->Failsafe mode -> Receiver ID -> bind -> Type (wrong order) .....

### Expected Behavior

Mode: FLYSKY -> AFHDS3 ->Type -> Module Options -> Channel Range CH1 ->CH18 ->Failsafe mode -> Receiver ID -> bind

### Steps To Reproduce

1. Model Settings
2. External RF
3. Walk with roller to different fields

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

philmoz, 2023-04-14:
The FlySkySettings class deletes and re-creates all the controls in the module options section (between Mode and Channel Range). This causes the order of the controls on the page to change when scrolling.

It needs to be rewritten to show/hide the controls instead.

I can look at this in the next couple of days if no one else is fixing it.

I also noticed that pressing the 'Module options' will crash the simulator.

richardclli, 2023-04-15:
> The FlySkySettings class deletes and re-creates all the controls in the module options section (between Mode and Channel Range). This causes the order of the controls on the page to change when [...]

Yes, please help to check.  Module options is updated by Flysky to bring setup features for PWM output for each channel.  You can check this PR to see what they have changed.  https://github.com/EdgeTX/edgetx/pull/3443

richardclli, 2023-05-05:
Reverted the PR, will work out the fix again.

PR #3479 (2023-04-18): fix(color): Scroll order for 'FlySky' module options
Fixes #3477 

Also fix simulator crash issues related to FlySky module.
closed 2023-06-19 by richardclli