EdgeTX/edgetx #3482: Emergency Mode after entering ELRS Lua

nangel0, 2023-04-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After flashing lastest nightly build radio powers and behaves as usual, accessing expresslrs Lua script OR tbs agent lite and turn Jr bay module wifi on crashes radio into emergency Mode.

Power cycling and try again it does it again, but it may take more or less time until emergency Mode, but never gets really into wifi mode

Ruled out:
module shorting (it behaves similar with several modules 
Bad battery 
Sd card

Downgrading to release v2.8.2 from 2023/04/01 everything seems to run ok (same 2.8 sd card , same files) 

### Expected Behavior

Turn on, do things without crash 

### Steps To Reproduce

Flash latest nightly
Sd card already on 2.8
Turn on
Access exlrs lua scrip (or tbs agent lite) 
Turn WiFi on 

EMERGENCY MODE 

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

Tx16s v1 (very first version) with updated v4 gimbals

richardclli, 2023-04-17:
Finally somebody got this.  I discovered it this weekend.  Confirmed both TX16S and EL18 having same problem. The condition is using 5.25M baud.

@nangel0 Please check if using a lower BAUD can resolve the problem.  I found that lowering the BAUD seems OK.

nangel0, 2023-04-17:
> Finally somebody got this.  I discovered it this weekend.  Confirmed both TX16S and EL18 having same problem. The condition is using 5.25M baud.
> 
> @nangel0 Please check if using a lower BAUD can resolve the problem.  I found that lowering the BAUD seems OK.
> 
> IIRC, the main should works sometime ago, maybe doing a bisect can help to nail down the problem.

Oh.. I got this one or two times some time ago, but I thought it was some kind of missed file from sd or something I messed up, but this weekend I got this behaviour very consistent, because I was working on happymodel es24tx and diy 2.4... I usually go out with r9 stuff and it works on 115k baud unlike the others...

Will dive in it further tomorrow...

richardclli, 2023-04-17:
I am doing a bisect to see what make it crash anyway.

richardclli, 2023-04-17:
SHA 8560c1  This is good
SHA e01659 This is bad

@raphaelcoeffic Any ideas?  You may need to play and wait for around 5-10mins for the radio crash and goes into emergency mode,.

richardclli, 2023-04-17:
I already nailed downed the change that cause the problem and revert it in a PR, and I figured out the merge has something missing.

nangel0, 2023-04-19:
> I am doing a bisect to see what makes it crash anyway.

On the latest master, as soon I select 1.87M baud and (for example turn on TX module wifi via LUA script ) it enters EMERGENCY MODE

115k didn't load LUA (~~conflict with previous air rate chosen?? was on 250HZ?~~ too low for 250hz)
400k OK
921 OK

richardclli, 2023-04-28:
> > I am doing a bisect to see what makes it crash anyway.
> 
> On the latest master, as soon I select 1.87M baud and (for example turn on TX module wifi via LUA script ) it enters EMERGENCY MODE
> 
> 115k didn't load LUA (~~conflict with previous air rate chosen?? was on 250HZ?~~ too low for 250hz??? per documentation it should work...)
> 400k OK
> 921 ~~OK~~ crashed after 10 min or so...

You used internal or external ELRS?  Which one did you used for testing?

raphaelcoeffic, 2023-04-28:
@nangel0 I suggest you re-test with the latest build from #3486. (use this with chrome/chromium: [buddy link](https://buddy.edgetx.org/#/dev/flash/pr?version=pr-3486%4053f172bb0703156db1cedaeb381bdbb88b8a9e9d))

nangel0, 2023-04-28:
> You used internal or external ELRS? Which one did you used for testing?

External ELRS module

richardclli, 2023-04-28:
> External ELRS module

Fixed and merged to head, please check again the nighties for sure.

nangel0, 2023-04-28:
> Fixed and merged to head, please check again the nighties for sure

Ibehaves the same way, sorry

check it here
[https://www.youtube.com/watch?v=zjoHzDigfn0&ab_channel=nellson.angelloFPV](url)

richardclli, 2023-04-29:
> > Fixed and merged to head, please check again the nighties for sure
> 
> It behaves the same way, sorry
> 
> check it here: https://www.youtube.com/watch?v=zjoHzDigfn0&ab_channel=nellson.angelloFPV

Which version did you use?  I need the SHA information.

nangel0, 2023-04-29:
> > > Fixed and merged to head, please check again the nighties for sure
> > 
> > It behaves the same way, sorry
> > 
> > check it here: https://www.youtube.com/watch?v=zjoHzDigfn0&ab_channel=nellson.angelloFPV
> 
> Which version did you use?  I need the SHA information.

Sorry, my bad, I flashed latest nightly (at that time) 

438fd0bd731978c19655d461635a6d3f23d76ca6

richardclli, 2023-04-29:
> Sorry, my bad, I flashed latest nightly (at that time)
>
> [438fd0b](https://github.com/EdgeTX/edgetx/commit/438fd0bd731978c19655d461635a6d3f23d76ca6)

You need an updated nightly, this is old version.

nangel0, 2023-04-29:
seems OK, have been on 53f172bb for 30 minutes handling LUA and several modules and everything is OK

commit 53f172bb (2023-04-28): fix: possible endless loop in case some error flag is set but RXNE is not

Fixes #3482
PR #3486 (2023-04-28): fix(usart): possible endless loop in case some error flag is set but RXNE is not

Fixes #3482

Summary of changes:
closed 2023-04-29 by raphaelcoeffic