EdgeTX/edgetx #3483: Use external module Sbus / PPM input mixed with internal module 4in1.

pierrotm777, 2023-04-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Describe the solution you'd like

I use Radiomaster tx16s MkII 4in1 and love it.
I should like to use an external module in same time than the internal module.
My idea would be to send Sbus or PPM signal in the external bay by using a DIY module and receive the signal to the internal 4in1 module.
Is it possible?

gagarinlg, 2023-04-16:
What exactly is your use case?
I did not fully understand what you want to do.

pierrotm777, 2023-04-16:
I friend has developed a module that send 16 command over a PPM or Sbus signal.
Behind a receiver we use a specific decoder witch convert these commands by on/off as Futaba Multiswitch worked. See here http://p.loussouarn.free.fr/arduino/exemple/BURC/BURC.html#encoder. But for use this Burc  coder/decoder system, we need to have an Sbus or PPM input.
It's my need.
We have developed other coder/decoder systems, by example this one.
https://github.com/Ingwie/OpenAVRc_Hw/tree/V3/MultiSwitch_Sw16-ProMicro.
This system is named X-Any and it's fully opensource. Burc and X-Any are developed by P. I Loussouarn (Rc-navy).

gagarinlg, 2023-04-16:
So, it is already possible to send S-BUS into the radio via the AUX connectors. You then get the channel values as trainer inputs. Just set one uf the serial AUX ports to SBUS Trainer and in the module set trainer mode to "Master/Serial". Them you have the option to use "TR1" to "TR16" as inputs for the mixers.
That way you do not need a second 4-in-1 module.
with the second 4-in-1 module you can use one as a redceiver for DSM, AFHDS2 and some other protocols, to send data to the radio, also for trainer setups, or whatever you want to do with the data.
The third option is to assigen channels 1-16 to one module and channels 17-32 to the second module, bind two different receivers to the modules and then have up to 32 channels on your model.

I am still not sure if this anserws your question.
If I understood correctly, you generate cahnnel data with an external circuit, with some special coding, and you want to send this into the TX16S to forward it to the model. If that is the case, just using the AUX port should be enough.

pierrotm777, 2023-04-16:
I know already the solution with the auxiliary input over the trainer system.
But I should like the same system with the external bay. I don't want use two receivers with 32 proportional channel. Our system use just ONE channel for send up to 16 commands for set on or off a fonction as sound, light. But with our system we can also use the old Multiswitch system, Futaba, Multiplex which run normally only with the old transmitter (41mhz, 72mhz) and not in 2,4ghz transmitter. With our coder and decoder, it's possible.

pierrotm777, 2026-04-03:
Hello,
Document 2.12 about Trainer isn't up to date, I don't see Master/PPM and Master/SBUS.
Trainer | EdgeTX User Manual https://share.google/8pgCHfu8bwdV2XBbZ