EdgeTX/edgetx #3490: Cannot build installer for EdgeTX companion in Windows 10 (MSYS2)

RomanVlg, 2023-04-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

All steps done from here https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Windows-10-%28MSYS2%29,
and all successful except one- building installer. (I've not using script all in one)

 The error is in the end of building windows installer
[image]

Beside Companion and Simulator doesn't start without installer or missing QT.
[image]

### Expected Behavior

Some mess in building path (QT)

### Steps To Reproduce

Just do step by step manual
https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Windows-10-%28MSYS2%29

### Version

Nightly (Please give date/commit below)

### Transmitter

Jumper T-Pro

### Operating System (OS)

Windows

### OS Version

11

### Anything else?

commit 54e0d35

RomanVlg, 2023-04-19:
So I reinstall all staff,  repeat manual  and it'all right

closed 2023-04-19 by RomanVlg

pfeerick, 2023-04-20:
Gotta love computers, eh? Glad it's working for you now. :)