EdgeTX/edgetx #3497: Internal Gyro on X-Lite Pro no longer working as Input

Gerd-St, 2023-04-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I reported a bug before for version 2.8.0,  "Heavily dampend gyro sensor on FrSky X-Lite Pro #2695" which is still in triage state since last November.
Today I compiled and installed the actual version from the main development branch for checking if the gyro sensor has been fixed.

The channel reaction is still very slow and dampend, if quickly tilting the transmitter from one extreme position to the oposite one it will still take several seconds until the channel output will follow.

### Expected Behavior

The channel output should follow the transmitter tilt angle without noticable delay.

### Steps To Reproduce

1. Create a simple model with one or two servo outputs, controlled by the TiltX and TiltY sensors
2. Go to the output monitor page and check the reaction to tilting the transmitter in X and Y directions

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X-Lite / S / Pro

### Anything else?

According to the version page on X-Lite my firmware was built at 2023-04-19 07:18, GIT# cf01c2b4.

Gerd-St, 2023-04-20:
Found a further bug regarding TilX and TiltY. If I will add a further mixer line for adding e.g. a rudder stick, the gyro sensor will no longer work at all!

[image]

pfeerick, 2023-04-20:
@JimB40 Did you say you had an X-Lite S/Pro?

raphaelcoeffic, 2023-06-12:
Duplicate of #2695

closed 2023-06-12 by raphaelcoeffic

Gerd-St, 2023-06-12:
Please also consider my comment from on Apr 20. So it is no longer just a duplicate of https://github.com/EdgeTX/edgetx/issues/2695.

pfeerick, 2023-06-20:
I've updated the initial post so this is different to #2695.