EdgeTX/edgetx #3500: Mechanism for Updating UI by INT/EXT Modules

richardclli, 2023-04-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

There are a few Issues required such a mechanism, I list out all of them here so that we can handle these together:

https://github.com/EdgeTX/edgetx/issues/3491
https://github.com/EdgeTX/edgetx/issues/2713
https://github.com/EdgeTX/edgetx/issues/3046

### Expected Behavior

After packet received by the modules, if the options screen is displayed, it need to be refreshed.

### Steps To Reproduce

Screen just not refreshed properly.

### Version

Nightly (Please give date/commit below)

### Transmitter

Flysky EL18, Flysky NV14, Radiomaster TX16S / TX16SMK2

mha1, 2023-04-22:
Please retest #2713 with #3504

richardclli, 2023-06-19:
This placeholder is now all finished.

closed 2023-06-19 by richardclli