EdgeTX/edgetx #3508: Add unit of measurement values on the Zorro telemetry screen

pfeerick, 2023-04-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When the telemetry from Multiplex is displayed on the RadioMaster Zorro 4-1 telemetry screen, there are no unit of measurement values after the numeric value.

[image]

### Describe the solution you'd like

It would be nice to have the unit of measurement after the numeric valie, for example 7.5**V**, 11.0**V** and 19.1 **°C** (or°F) and other possible units such as **A** for current etc.

### Additional context

Originally incorrectly submitted as #3507

3djc, 2023-05-02:
Issue is screen resolution, not much miracles can be done on 128x64 pixels displays

pfeerick, 2023-05-03:
Would first need to consider what the longest sensor name can be, and then how many digits the sensor values can go up to.

mha1, 2023-05-04:
Can sensor names have more than 4 characters?

pfeerick, 2023-05-04:
I just checked, and *at the present time*, no. However, the reason for that limit is no longer present, so that may change in the future.

Even so, when making UI changes, you need to consider the largest values, and work backwards...

[image]

While I think the Heading is a bit silly with two significant digits after the decimal, that is what it defaults to. 

Basically, I don't think units can just be added, with that particular screen layout. It would probably have to be a single column, not two columns, then there would be enough space to work with for all sensors without losing any precision.

M-i-c-h-a-e-l1, 2023-05-05:
Ok, I get the point. I only have Multiplex telemetry sensors for temperature and battery voltage as this is for a trial truck. As shown on your picture above, this will probably not work if you have telemetry sensors which have more 4 or 5 digits. 

A work around could be, as you mention, to only have one column with 3 or 4 values. The Radiomaster Zorro can have up to 4 independent screens with telemetry values. This would however mean that one would manually press the 'Telem' button on the Zorro radio control to select the next telemetry screen. This is also done in other radios when there is not enough space on the screen for all telemetry values.