EdgeTX/edgetx #3515: EL18 Vspd in telemetry screen

CharlesGauci, 2023-04-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Would it be possible to include 'Vspd' with 'Alt' and 'Temp1' in the telemetry screen of the EL18 similar to Pull Request numbers NV14 #1477 https://github.com/EdgeTX/edgetx/issues/1477 and NV14 #1485 https://github.com/EdgeTX/edgetx/pull/1485 ?

### Describe the solution you'd like

It should make vario function work if implemented

richardclli, 2023-04-28:
Well, the firmware for NV14 and EL18 are nothing different except the name displayed.  So what works with NV14 should works with EL18 as well.

CharlesGauci, 2023-04-28:
Thanks for the reply richardclli , yes I am on 2.8.3 and I use both NV14 and EL18, but Vspd only appears on NV14.

pfeerick, 2023-05-01:
#3386 is the actual relevant PR changes stem from, #3467 is simply a backport of it with no functional changes. 

I commented there that the altitude, which is used to calculate VSpd on the NV14/AFHDS2A, that:
> I'm pretty sure the previous behaviour was coded so that it would be relative to takeoff, i.e. 0 and then go up or down. whereas now for me it starts off at about 48m and goes up in 0.5m increments. Is it supposed to be an approximation relative to sea level or something?

https://github.com/EdgeTX/edgetx/pull/3386#issuecomment-1509481649

Rather than using this altitude code like before:
https://github.com/EdgeTX/edgetx/blob/53f172bb0703156db1cedaeb381bdbb88b8a9e9d/radio/src/telemetry/flysky_ibus.cpp#L517-L554

it now uses this code, with a table of values in 0.5m increments and references to sea level presssure?

https://github.com/EdgeTX/edgetx/blob/53f172bb0703156db1cedaeb381bdbb88b8a9e9d/radio/src/telemetry/flysky_ibus.cpp#L555C4-L618

pfeerick, 2026-11-12:
> Well, the firmware for NV14 and EL18 are nothing different except the name displayed. So what works with NV14 should works with EL18 as well.

After some digging, that isn't actually the case... depending on the RF firmware version of the NV14, it takes one of two different paths when fed the altitude telemetry value, and only the one for the older rf firmware created the VSpd sensor. On the newer RF firmware, it behaves the same as the EL18 and MPM, and doesn't create a VSpd sensor. See https://github.com/EdgeTX/edgetx/issues/3847#issuecomment-6814296792 for more detail. #8036 should go some what to resolving this.

closed 2026-11-12 by pfeerick