EdgeTX/edgetx #3518: Model Select: lastOpened doesn't get saved to labels.yml correctly

ducatiJake, 2023-04-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In Model Select, if I have sort order set to last opened at top of list, while I stay powered up I can switch models multiple times and the last opened models are always in proper order towards the top of the list.  If I power down and restart Edge then try to select a model, the current model is first, but my previously selected models no longer follow as they did before I powered off.  I looked in the labels.yml file on the radio and I have never seen more than 2 models with lastopen set to anything other than 0.  For example, if I have model A selected on power up, then select model B, then open model select again, I will see model B selected and model A positioned as the next model.  So cancel out of model select.  Now power down and then restart.  Now open select model and model B will again be shown as selected, but model A is no longer positioned as the next model as it was before the shutdown.

### Expected Behavior

I would expect that the model order displayed in select model would be the same across power shutdowns if i have made multiple changes to the selected model.  As stated above this seems to be because lastopen in the labels.yml file does not seem to be being stored properly.

### Steps To Reproduce

See Current Behavior for steps.

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

philmoz, 2023-04-24:
The 'last opened' value is written to the 'labels.yml' file correctly,

The problem arises because the value is not being read correctly from the 'labels.yml' file on startup.

The hash values used to determine if the model entries in the 'labels.yml' file are valid are not updated when the model files change. On the next restart the 'last opened' values are ignored because the hash does not match.

ducatiJake, 2023-04-28:
Tried this workaround it seems to work fine.

PR #3521 (2023-05-07): Workaround for 'lastopened' value not loaded on startup if stored hash not up to date.
Fixes #3518 

The hash value stored in the labels.yml file does not get updated when model files are saved.
On restart the values stored in labels.yml are ignored because the hash is not correct.

This workaround forces loading of the 'lastopened' value regardless of the hash. All other ignored values are re-created correctly.

This is a quick and dirty workaround to fix the issue.
closed 2023-05-07 by pfeerick