EdgeTX/edgetx #3527: the charge indicator on the 6-position switch should flash

gumileeev, 2023-04-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have jumper t pro. The lights on the 6 position switch indicate the amount of charge. The problem is that there is no difference between starting a charge when the number six light comes on and completing a charge.

### Describe the solution you'd like

It would be great if each bulb, when it is on the far right, blinked. Those. when the 6th division has just begun, it blinks. when the charging process is completed - the light is just on.

### Describe alternatives you've considered

The jumper t pro manufacturer's instructions indicate that the green LED next to the USB port should be on during charging. When charging is complete, the green LED should turn off. This does not work, the green LED is always on. Therefore, there is no way to understand by indication when charging is really over.

gumileeev, 2023-04-24:
The end of the charging process can be considered the achievement of the set end voltage, which is indicated in the settings.

pfeerick, 2023-04-25:
FYI, the voltage set in the settings is not for charging, only for the UI - for things like the battery guage. Firmware/EdgeTX has no control over the charging process - this is managed by dedicated charging IC.

gumileeev, 2023-04-25:
> 

I'm not asking you to interfere with the charging process. I want the indicator to blink the rightmost light while charging. If 3 lights are on, then the third light is flashing. When the voltage has reached 8.2v (I don’t know when exactly it lights up, this is for example) - the sixth light starts flashing, and when the voltage has reached 8.4v (the value that is already in the settings that affects the indication), then the sixth The light stops blinking and stays on all the time. Then we conclude (conditionally) that the charging process is completed (although in reality, the charge controller can, for example, continue to do something, maintain the charge, we do not affect charging, we only make the indication clearer)

gumileeev, 2023-04-25:
I am attaching a video with the process of charging a device with a charge indicator consisting of 4 bulbs. It shows that the charging process is still in progress. Now on the "t pro" the light just lights up, it's not clear, or the light just turned on, and the charging process is still going on. Or the light bulb has been on for a long time, and the voltage has already reached the extreme value from the "radio setup - Batt. range" settings. A blinking indicator would make the charging process very clear.
https://youtu.be/MbVNitVhwIo

gumileeev, 2023-05-02:
If it is not possible to manage indicators, please close my request.  Apparently, the battery setting is applicable only for indication within the program, but not for LEDs.

3djc, 2023-05-02:
There is no proper way for the the software to know the charge status, as this is  managed by a external IC thats not conected. What is currently done is actually starting the processor and turning on part of the radio systems to measure the voltage value. Measuring the voltage is 1) not precise enough 2) does not allow to properly define charge end for li-xxx type devices

pfeerick, 2023-05-02:
I never did check, what is the charge chip in these - TP4050? If so, the charge cycle on these can also be weird - it will actually charge up to 100%, let it run back down somewhat and then charge up again if the battery has a load on it while charging - which makes it good for projects where the battery that is "always charging" since it never gets held at 100% charge.

3djc, 2023-05-02:
i have asked, and will suggest they wire it to MCU in the future, so we have some kind of charge done info

gumileeev, 2023-05-02:
> I never did check, what is the charge chip in these - TP4050? If so, the charge cycle on these can also be weird - it will actually charge up to 100%, let it run back down somewhat and then charge [...]

[image]
[image]

rgvfpv, 2024-04-04:
Was this ever resolved? I have a Jumper T -Pro V2 and I love this remote but it does bother me that I can leave it charging and there is no difference between the start of the last light turning on and full charge. Why not just shut off all lights when it's fully charged? I believe the Jumper T14 does that.

pfeerick, 2024-04-04:
I would be surprised if the T14 does... The code simply sets the led color
in three stages, with that latter simply being if above a certain value. If
you are referring to the white led, that is hardware controlled (like the
green one on the TPro mentioned above). The TX16S is similar in this regard
- the charge indicator is hardware, not software, controlled.

Given that as previously discussed the issues of calibration and no actual
knowledge of if the charge process is complete, I don't think we can
reasonably do much more than what we have now... anything more will be
disingenuous since.

If you want a better idea of the charge termination (and can't just leave
the radio plugged in to charge overnight), your best bet is to use a charge
cable with current monitoring, or a USB power meter, and wait for the
charge current to drop off.

closed 2024-04-04 by pfeerick