EdgeTX/edgetx #3528: Switch to C++20 standart

ggoraa, 2023-04-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I would want to use modern C++ features in the EdgeTX codebase, like std::format, but they are not available because of the C++ standart set to C++11

### Describe the solution you'd like

Well, switch to C++20

### Describe alternatives you've considered

~~switch to Rust~~

pfeerick, 2023-04-26:
You might want to read this for some context. https://github.com/EdgeTX/edgetx/pull/1853#issuecomment-1108605218

You're more than welcome to port ETX over to Rust... someone has already ported it over to the ESPIDF... so it shouldn't be too hard for you whip in up in a couple of weekends 😛 https://www.rcgroups.com/forums/showthread.php?4327269-EdgeTX-running-on-ESP32

ggoraa, 2023-04-26:
Understood, so there is a goal to move over to C++20(or C++17 if C++20 is too big of a jump), it's just that it's not high in the list of priorities, correct?

gagarinlg, 2023-04-26:
There is more to it, like Compiler versions, Qt for companion and simulator and so on. Currently the main build platform is Ubuntu 20.04 which does not fully support C++20

ggoraa, 2023-04-26:
Is C++17 a good alternative?

elecpower, 2023-04-27:
Qt 6, which we plan to move to, currently only supported up to C++17 and Ubuntu 20.04.

elecpower, 2023-04-27:
Update: Just rechecked and above applies to Qt 6.4 and earlier however Qt 6.5 supported to C++17 and Ubuntu 22.04.

Note: Qt5 supported to C++11 and Ubuntu 18.04 and is the last support for 32 bit Windows which we still have lots of downloads.

There is a plan for 2.10 to push 32 bit support to the latest Qt 5 to maximise its life but Qt is obsoleting functions and features as it progresses with Qt 6.

ggoraa, 2023-04-27:
I think a lot of people that download 32bit versions of companion are confused, because I don't think that anyone now has a 32bit system and OS

elecpower, 2023-04-28:
In the past pilots have reported using an old clunker laptop they take to the field to run Companion but as you say others don't know what they are running.

jtaylor2, 2023-04-28:
There is at least one person that still has a 32bit system and OS.  I have a ASUS T100TA Transformer that I use at the flying field running 32 Bit Windows 10.  The machine itself is 64 bit but it came with a 32 bit OS because it doesn't have enough memory (non upgradable) or SSD (non upgradable) to run 64 bit Windows.

While Microsoft ended support for 32 bit Windows 10 on new machines 3 years ago in May 2020 I don't think it goes out of support until Fall 2025.  I know I'm still getting updates for mine.

I don't know how many people are like me and use an old low end machine at the field instead of  my good one, but I doubt I'm the only one.