EdgeTX/edgetx #3529: Sound choppy after unplug from USB mode

phileville, 2023-04-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have several switches that are configured to plan sound files when they are switched.  So when I load a model, it will play an audio file "disarmed", "flight mode Acro", etc based on the swith positions.  When I unplug my USB Cable after being into USB mode, The radio goes plays the sounds again like a new model is loaded, except in this case, the sound is played choppy, like the processor is having a hard time keeping up... I only noticed this when I updated to the nightly form 23 April.   

### Expected Behavior

Sounds should play smoothly

### Steps To Reproduce

Configure a play track special function with a 1x repeat and assign it to a switch.
Make sure that the switch is enabled (so the track will play)
Plug in USB cable and select USB mode
Unplug USB cable and configured sound will play, but it will be choppy.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

mha1, 2023-04-28:
I followed your procedure with tx16s-438fd0b.bin (dated 04/23/2023) but can't reproduce the problem. Can you please confirm the nightly commit version by checking SYS/VERSION output.

phileville, 2023-04-29:
Sorry, maybe my steps to reproduce were not clear enough..  I made a video to show it.  I flashed to the nightly from today (53f172bb) and it still does it - here is the video link - https://www.youtube.com/watch?v=CIViE2qPcN0

mha1, 2023-04-29:
Your instructions were quite clear. Here's a video of an attempt to recreate your issue with fw version nightly 53f172b and the same audio files on the same switch you used: https://youtu.be/fZdeKL3KE3Q. I can't reproduce the issue.

Thinking loud:
- do you have LUA scripts running in the background. Check if you have LUA Script entries in SFs or GFs. If so disable them one by one in order to try and find the offending one. 
- maybe the horizon audio file is corrupt. Here's mine to try: [fm-hor.zip](https://github.com/EdgeTX/edgetx/files/11359852/fm-hor.zip)
- maybe the SD card has a problem. Try to check for file system errors with PC tools e.g. [Windows](https://www.computerhope.com/issues/ch001118.htm)
- try different SD card

phileville, 2023-05-05:
HI - So I did some more testing, multiple SD cards, new SD card contents, new models from scratch.  It seems the problem stems from when you have a special function or global function that is configured to play a track and the model is loaded and the switch to play the configured track is active so the track should play by default.  Simply switching models is enough to create the occurance - no need to go into USB mode.  

New recreate steps - 
1. create a new model
2. configure a special function "play Track", select any one of the sounds provided on the normal sd card soundpack. I am using "disamed" - note you can also configure it to "play value" for one of the axes and it will also do it. Make sure the repeat is set to 1x and that the trigger switch is set and in the position to trigger it. 
3. Go to manage model screen and set a different model to active.
4. Go to manage model screen again and set the newly created model to active.
5. You should hear the tracks or announcements played and they will be choppy -  note: if you have more than one SF configured, then you have a longer period to hear the choppiness.
Also note, that after the initial loading of the model and announcements, the tracks will play totally fine when activated by the switch.  It is only doing this when the model is loading up.

I hope that this is enough for you to recreate the issues.  If not, then I am not sure what else to say.  It is not so big of a problem, but just a little annoying.