EdgeTX/edgetx #3530: Companion Simulator core dump

icebreaker-ch, 2023-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

System: Linux Mint
Application and Version: EdgeTx_Companion_2.8.2-x86_64.AppImage
Problem:
Opening the simulator for the model "T-Thermic-E bg" in the attached configuration
[Horus X10S ETX.zip](https://github.com/EdgeTX/edgetx/files/11342541/Horus.X10S.ETX.zip)
crashes the simulator.

Output on terminal (last few lines)
```
0.35s: f_closedir(0x7f8b99dfb4c0)
0.35s: f_open(/home/sro/SynologyDrive/Data/EdgeTX/Horus X10S/SDCARD/THEMES/selectedtheme.txt, 1) = 0x7f8b8807f080 (FIL 0x7f8b99dfb900)
0.35s: f_close(0x7f8b8807f080) (FIL:0x7f8b99dfb900)
0.35s: f_open(/home/sro/SynologyDrive/Data/EdgeTX/Horus X10S/SDCARD/THEMES/EdgeTX/background.png, 1) = 0x7f8b8807f080 (FIL 0x7f8b9be9c920)
0.36s: f_close(0x7f8b8807f080) (FIL:0x7f8b9be9c920)
0.36s: f_open(/home/sro/SynologyDrive/Data/EdgeTX/Horus X10S/SDCARD/THEMES/EdgeTX/background.png, 1) = 0x7f8b8807f080 (FIL 0x7f8b9be9c920)
0.36s: f_close(0x7f8b8807f080) (FIL:0x7f8b9be9c920)
0.36s: opentxStart(7)
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Aborted (core dumped)
```

### Expected Behavior

Simulator should open without crashing

### Steps To Reproduce

1. Extract the file HORUS X10S.etx from the attached ZIP-File
2. Open the AppImage
3. Create a radio profile for HORUS X10S
4. Open the file HORUS X10S.etx
5. Open the Model called "T-Thermic-E bg"
6. Start Simulation
7. Simulator core dumps

### Version

2.8.2

### Transmitter

FrSky X10 / X10S (ACCST)

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.1 Cinnamon

mha1, 2023-04-27:
Fixed in 2.8.3. Please retest with Companion 2.8.3 and close this issue if it solves your problem.

[image]

icebreaker-ch, 2023-04-27:
Works correctly in v2.8.3. Thanks a lot for fixing.

closed 2023-04-27 by icebreaker-ch