EdgeTX/edgetx #3531: Apply Labels to Global Functions

jmxp69, 2023-04-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Certain features available as a global function such as instant trim and play track make perfect sense as global functions; however, there may be a subset of global functions one might wish to have removed from certain models while retaining other global functions.

### Describe the solution you'd like

Let's use these global functions as an example:

ON Volume S1
1) SBup Play Track High Rates
2) SBup Play Track Bank One
3) SCdown Play Track Stabilizer Off
4) SHdown Instant Trim
5) 6P1 Angle
6) 6P2 Acro
7) SAup IdleUp2

On the following model categories, I would like all of these global functions:

Fixed Wing No Stabilizer: 1, 4
Fixed Wing Stabilizer: 1,3,4
Helicopter: 2, 7
Quad: 1,5,6

Using Model Labels, I would make categories: FW, FWStab, Heli, Quad

Global Functions could then use: ALL, FW, FWStab, Heli, Quad for each function so that when a user applies a label to a model the associated global functions are applied. 

This would extend the use of labels and give a degree of flexibility to how global functions are used to avoid the binary on/off option which exists today. Usually, I don't want to turn global functions off; however, I also don't want SAup to sound IdleUp2 on a fixed wing model.

### Describe alternatives you've considered

Obviously the alternative is moving global functions that aren't "truly" global to the Special Functions tab but this makes model setups more complicated than they need to be if we can categorize models to use select global functions.

### Additional context

Looks like I picked a bad week to quit huffing paint.

pfeerick, 2023-07-06:
I really don't know about this one... I know what is trying to be achieved - functions that can be shared by groups of models allowing for changes to be made in one place rather than per model - but I'm not sure this is the right way to do it. I think either something more along the lines of what Malte said is the right way to do it, or perhaps something completely different (some sort of template functions - that can be matched on a label or a per model basis), leaving GFs exactly as they are - since by definition they are global. I think it would be more confusing at this point to redefine GFs and say they are global, except for, except for. 

Personally, I only have a couple of global functions, most of which should be going the way of the dodo shortly, moving properly into system settings rather than being global functions. It never made sense to me to use global functions for more than radio  functions like brightness, volume, screenshot, touch lock, etc, as I don't use one type of RC craft, so what suits one category invariably doesn't suit any of the others - i.e. never used instant trim for cars, boats, quads or helis, so it makes no sense to set as a GF. But I can see that in some circumstances, i.e. if you only fly planes, sure, it makes sense. What would be nice to have is a central bank of functions that can be used by selected models, or shared by a category. However, I see it more being needed at the model level of things, as I want to be able to use the same logic switches, the same special functions, same GVs. When up update my throttle logic and associated SF callouts, etc,  I'd like it usable for all the other models in that same category.

jmxp69, 2023-07-24:
I'm all for more discussion and I like some of the ideas you tabled like a bank of functions selectable by model function.

The thing that gets me into trouble is my use of the left slider which I usually define as a global function to adjust brightness. This works GREAT until I have retracts which I also like to put on LS. So for those planes, I'd like to not see my brightness go down when I put my gear down. I use S1 for volume and S2 for gain. An S3 would be really helpful!

I also like the idea of logic that can be applied to groups. i.e. Adding L01 AND TELE SHup as my new SD card log trigger and my timer trigger is brilliant. Going back to 35 models and adding it kinda not so fun. And this happens to me quite a bit as I experiment and come up with new ways to do things--I usually wind up spending quite a bit of time in Companion.

Trainer is another good example: Setting up a global trainer config then being able to just select the models to which it applies would be absolutely brilliant rather than going model by model and inserting the trainer commands into each configuration.

i.e.:
Flight Mode 1,2,3 - Trim 5 disabled
L01 Sticky T5d T5d
SF1 L01 Play Track Master Control
SF2 !L01 Play Track Student Control
SF3 !L01 Trainer Chans Enabled
System - Aux1 SBUS Trainer Port Power On

It would be so so much nicer to build this logic out and just say: Apply this to Model XYZ so I can then quickly add a trainer setup to anything and let my friend fly my plane on the buddy box.

Same deal with throttle lock. Dozens of ways to do this. It would be great to define a throttle lock in one spot then just apply throttle lock to whatever models.

This would drive a whole new level of consistency across radios with a lot of models...

pfeerick, 2023-07-26:
Maybe have another screen, "Snippets"/whatever, and on that screen you have the all blocks that would be available for inclusion for each model. You can enable/disable the blocks active for the current model from there. You would then go into each block, and be able to edit the GVs, LS, SF, etc contained in each... That makes the UI ... *easy* ... I guess... making it actually work would be the fun bit.

jmxp69, 2023-07-27:
That would work for me. Just a label to identify what that block does and some way to attach it to any given model or even models with a specific label..?