EdgeTX/edgetx #3534: TX16S Hang with Sound+Vibration Loop when Switching Models with Rx Connected

ssvv123, 2023-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When a quad is connected to my Radiomaster TX16S radio, and I attempt to change to a different model profile via the touchscreen, the radio will refuse any new inputs, and begin vibrating constantly while repeating "receiver still connected" until the battery is pulled. After reinserting the battery pack and powering back up, it will be in Emergency Mode until power cycled again.

### Expected Behavior

The radio will change to a different model profile seamlessly without throwing an error, or will prompt to override.

### Steps To Reproduce

1. Power on radio with external ELRS module
2. power on quad which is bound with radio
3. tap once on touchscreen to show menu
4. tap on model select button
5. tap on any other model profile than the currently active one
6. brace yourself

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

This is using an external ELRS module (Happymodel ES24TX Pro)
No Model ID is set for any of my profiles
This has been reproduced on a Mobula 7 1S w/ SPI ELRS and a 5" SpeedyBee DIY with Happymodel EP2 Rx
This happens every time these steps are followed without variation

ssvv123, 2023-04-27:
Video of bug can be found at https://vimeo.com/821821520

philmoz, 2023-04-27:
This should be fixed in 2.8.2. Please give 2.8.2 or 2.8.3 a try,

ssvv123, 2023-04-27:
Whoops, you're correct! 
Fixed by flashing 2.8.3 

Thanks <3

closed 2023-04-27 by ssvv123