EdgeTX/edgetx #3535: program for the new jumper t pro v2

gumileeev, 2023-04-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

there was a jumper t pro v2, it has a new processor, a new display, additional switches. Will there be an adaptation for the new version?

### Describe the solution you'd like

As I understand it, you can’t just take and update the version from jumper t pro v1, because. different processors and different display driver. Is it worth waiting for them under a new jumper? Can the owners of the new jumpers do something to help?

gumileeev, 2023-05-02:
Sorry, didn't see that the new t pro v2 comes already with 2.9.0

closed 2023-05-02 by gumileeev

pfeerick, 2023-05-02:
There was some confusion as to the release of this before firmware was ready for this, so it will most likely be necessary to run a nightly for 2.10 once https://github.com/EdgeTX/edgetx/pull/3554 is merged, as the firmware that comes on the radio is not fully functional, nor is it an official release from EdgeTX.