EdgeTX/edgetx #3551: Incorrect display of the crosshair vertically

Shouldercannon, 2023-04-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When checking the touch screen operation, a crosshair showing the place of touch is displayed.
The crosshair vertically is not displayed correctly.
[image]

### Expected Behavior

The crosshair should be horizontal and vertical together touching.

### Steps To Reproduce

1. Push Sys and go to Hardware menu.
2. Go to the bottom of the Hardware menu and select Analogs.

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Win10 x64 22H2

### Anything else?

This problem has been observed since version 2.8.0 and was inherited until version 2.8.3.

ggoraa, 2023-04-30:
one of those rare cases when debug helpers are the ones that actually malfunction

philmoz, 2023-04-30:
This is fixed in PR #3240 which I believe is scheduled for the 2.9 release.

pfeerick, 2023-05-01:
Indeed it is - #3240 is merged so it's present in the nightly builds leading up to 2.9

closed 2023-05-01 by pfeerick