EdgeTX/edgetx #3552: Sound files can not be browsed

ulfhedlund, 2023-04-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have quite a few sound files (approx 230 in the "en" folder) on my SD card and I have had a few sound related issues. 

When I navigate to the SOUNDS/en folder I get only four items listed: SYSTEM, 100.wav, 1000.wav and 10sec.wav. Sound folders with less number of files (yaapu for instance) can be browsed without problems. 
Sounds already assigned to special functions play as expected but they can not be changed in the Model Settings/Special Functions/Play Track. Nothing visible happens when I click the folder icon in the Value field. 

Similar issues also occur in the 2.8.3 simulator, although not identical: clicking on the Special Functions - Play Track - Value field (folder icon) freezes the simulator (and on some occasions also Companion). After navigating to the SOUNDS folder nothing happens when selecting any of the folders or when clicking on the .. (up one level). 

On my Xlite running 2.8.3, with the same sound files on the SD card, everything works as expected.

### Expected Behavior

All sound files can be browsed even though there are many files in a directory.
Any existing sound file can be selected in Special Functions and assigned to a Play Track SF.

### Steps To Reproduce

See above

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Linux

### OS Version

Kubuntu 22.04

### Anything else?

Firmware 2.8.3, SE language build
Companion for Linux (EdgeTx_Companion_2.8.3-x86_64.AppImage)

philmoz, 2023-05-01:
My TX16S has no problems with 234 files in the sound folder.

Have you checked the SD card for errors?
I would suggest trying a different SD card, or copy all the contents and re-format the card.

ulfhedlund, 2023-05-01:
I have 874 files in the "en" sound folder, not sure if that makes any difference. Will do some more testing as well as trying a brand new SD card and report back.

ulfhedlund, 2023-05-01:
I created a new SD card and copied all files from my hard drive.
However the issues persist. I still can not browse the files in the SC card SOUNDS/en folder. Out of the 874 files (way too many, I know), only the first four entries are listed. And I still can't select any English sound file in the Special Functions - Play Track - Value field. 

However, if I change language to "se" (Swedish) I can browse all 230 Swedish sound files and select any of them to play in the Special Functions - Play track. So from my tests so far the issue seems to be related to the large number of sound files on my CD card.

philmoz, 2023-05-02:
This appears to be a limitation in the LVGL lv_table object used by both the file browser and the popup menus (when you select a sound for a special function). The limit seems to be a combination of the number of rows and the row height in the table.

At the moment the limit looks to be 803 rows, so you are limited to 800 sound files (plus the '..', 'SCRIPTS' and 'SYSTEM' folders).

If you want to build a custom firmware you can also decrease the row height for the table cells which will allow more rows.
Add the line below to the radio/src/gui/colorlcd/themes/etx_lv_theme.cpp file after line 609.
    lv_style_set_max_height(&styles.table_cell, 35);

gagarinlg, 2023-05-02:
> 

So it is more a limitation of the maximum height of a single object?

ulfhedlund, 2023-05-02:
I decreased the the number of files to less than 800, which solved this issue just as explained by philmoz.

ripperhack, 2024-03-27:
I have the same problem too. Radio boxer, latest version of edgetx. From the sys menu I can browse and play the sounds in SOUNDS/en.
From the special functions play sound the system says "no sounds on sd card".
I tried leaving only 2 files in SOUNDS/en but the result doesn't change

pfeerick, 2024-03-27:
Is SYS -> Radio Setup -> Voice Language is set to 'EN'?

https://manual.edgetx.org/v/v2.9/b-and-w-radios/radio-settings/radio-setup
> Voice language - Language for the voice pack. This setting and the voice pack folder on the SD card must match for the sounds to be played.

And is the path is exactly `/SOUNDS/en`, i.e. `/SOUNDS/en/acro.wav`, `/SOUNDS/en/SYSTEM/0000.wav`?

B&W is a completely different beast to colorlcd, so it won't be the same problem here.

ripperhack, 2024-03-27:
sorry, you're right, I didn't read the radio model. However the language is correct in the sys menu and the path is exactly as you suggest. It doesn't matter... I'll live without having audio :)
thx @pfeerick

pfeerick, 2024-03-27:
Hm... okay... so 

- path is correct
- voice language is correct
- sound actually works if you play the files via sd browser
- SF "Play Track" says there are no sounds on the card (implying `/SOUNDS/en` is empty)

Since you get sound, no need to check the speaker connection or volume settings... that's a new one... 

What version of ETX... 2.9.4, 2.10.0-rc1, something else? And just double checking... it's `/SOUNDS/en` etc, not `/SOUNDS/SOUNDS/en`?

ripperhack, 2024-03-27:
exactly, everything is correct (checked 20 times). EdgeTX "Providence" version v2.9.4

I think it's an SD problem (new, 64Gb, fat32, formatted).

Copied / pasted from old to new with pc mac

This evening I'll try to do it again with another SD (32Gb) from a Windows PC

I'll update tomorrow

3djc, 2024-03-27:
> I have the same problem too. Radio boxer, latest version of edgetx. From the sys menu I can browse and play the sounds in SOUNDS/en. From the special functions play sound the system says "no sounds on sd card". I tried leaving only 2 files in SOUNDS/en but the result doesn't change (no other folders inside SOUNDS, only en)

Also play sound doesn't actually look for sd card files since it plays synthetized frequencies with choices like beeps, maybe you meant play track ?

pfeerick, 2024-03-27:
Hm... very strange... if you can play them via the sdcard browser, it shouldn't even be that really, as the radio can clearly read the sd card (it has to be able to, otherwise it basically wouldn't work).

3djc, 2024-03-27:
> Hm... very strange... if you can play them via the sdcard browser, it shouldn't even be that really, as the radio can clearly read the sd card/files (it has to be able to, otherwise it basically wouldn't work).

Could be naming

3djc, 2024-03-27:
> my idea is that by doing the sd with mac pc it automatically added hidden folders or something with spotlight

While true (the hidden files), it poses no issue

ripperhack, 2024-03-27:
Tomorrow I'll update you on the test I'm doing this evening with a new SD card made without copying/pasting and let's see what happens

ripperhack, 2024-03-28:
Problem solved, new SD with Windows PC. Copy/paste from the old one.
Tried again with the old one made with Mac, it doesn't work

pfeerick, 2024-03-28:
ouch! Well, at least it works now. :smile:

marcosx86, 2025-03-04:
Is there a way to browse and play any sound file from SDcard using special functions?