EdgeTX/edgetx #3557: OTG video using display

robertmeisner, 2023-05-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The current EdgeTX firmware does not support  displaying video receiver's (like skydroid) USB On-The-Go (OTG) video on the built-in LCD screen of transmitters such as the Radiomaster TX16S. This limits the usability of the transmitter for pilots who rely on real-time video feed, such as FPV drone pilots or other RC hobbyists, requiring them to use additional equipment or workarounds to view the video feed.

### Describe the solution you'd like

I propose the implementation of a new feature in EdgeTX firmware that enables the display of OTG video on the LCD screen of supported transmitters, such as the Radiomaster TX16S. This feature would allow users to view real-time video feed from their aircraft or vehicles directly on their transmitter's screen, eliminating the need for separate display equipment and enhancing the overall user experience.

### Describe alternatives you've considered

An alternative solution could be the development of a separate module or DIY add-on hardware that connects to the transmitter and allows for the display of OTG video on the LCD screen. However, this would increase the overall cost and complexity for users, as they would need to purchase and install additional hardware. Integrating this feature directly into the EdgeTX firmware would provide a more seamless and user-friendly solution.

### Additional context

The addition of OTG video display support in EdgeTX would greatly benefit the FPV and RC community, as it would provide an all-in-one solution for pilots, reducing the need for additional equipment and streamlining their setup. This feature would also make the Radiomaster TX16S and other supported transmitters more versatile and attractive to a broader range of users.

pfeerick, 2023-05-03:
I'll defer to the rest of the dev team, but to the best of my knowledge
this cannot happen as I don't believe we can operate the USB in host mode,
nor can the screen be driven fast enough for video while also functioning
as a transmitter...

gagarinlg, 2023-05-03:
The USB ports in the radios are not wired to support OTG. For the radios with USB C a special IC would be needed to convert the USB C mode to OTG Mode and those are not present. Also, I think, that the FPV equipment does not usb USB but Display port or hdmi over the additional pins on the USB cable, which the radios definitely can not support. 
The MCU of the radios is too slow to handle a video stream and rescale it to fit into the screen and the screen framerate is too slow to play real time videos.
There is no multi GHz Cortex A in those radios, like in a cell phone, but a 1xx MHz Cortex M, like in the flight controllers.

closed 2023-05-03 by gagarinlg