EdgeTX/edgetx #3558: MSYS2 setup and firmware build warnings

mha1, 2023-05-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

Don't know if actual issues or to be expected. I'm just curious.

I installed MSYS2 build system on a new machine. and saw some warnings in setup stage 2 like this one:
```
C:/msys64/mingw64/lib/python3.10/site-packages/setuptools/command/build_py.py:201: _Warning: Package 'pyppmd.cffi' is absent from the `packages` configuration.
!!

        ********************************************************************************
        ############################
        # Package would be ignored #
        ############################
        Python recognizes 'pyppmd.cffi' as an importable package[^1],
        but it is absent from setuptools' `packages` configuration.

        This leads to an ambiguous overall configuration. If you want to distribute this
        package, please make sure that 'pyppmd.cffi' is explicitly added
        to the `packages` configuration field.

        Alternatively, you can also rely on setuptools' discovery methods
        (for example by using `find_namespace_packages(...)`/`find_namespace:`
        instead of `find_packages(...)`/`find:`).

        You can read more about "package discovery" on setuptools documentation page:

        - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html

        If you don't want 'pyppmd.cffi' to be distributed and are
        already explicitly excluding 'pyppmd.cffi' via
        `find_namespace_packages(...)/find_namespace` or `find_packages(...)/find`,
        you can try to use `exclude_package_data`, or `include-package-data=False` in
        combination with a more fine grained `package-data` configuration.

        You can read more about "package data files" on setuptools documentation page:

        - https://setuptools.pypa.io/en/latest/userguide/datafiles.html

        [^1]: For Python, any directory (with suitable naming) can be imported,
              even if it does not contain any `.py` files.
              On the other hand, currently there is no concept of package data
              directory, all directories are treated like packages.
        ********************************************************************************

!!
```

The build system works but throws a warning at the end I hadn't seen before:
`
c:/msys64/mingw64/bin/../lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld.exe: warning: ../../firmware.elf has a LOAD segment with RWX permissions
`

### Expected Behavior

don't really know what to expect. No warnings? At least I'd like to understand the firmware build one.

### Steps To Reproduce

Do a clean MSYS2 build system setup as per Wiki.

Build firmware using script:
```
cd <path-to-local-repo>/EdgeTX
mkdir -p build-output-main-tx16s && cd build-output-main-tx16s
#
cmake -G "MSYS Makefiles" -Wno-dev -DCMAKE_PREFIX_PATH=$HOME/5.12.9/mingw73_64 -DSDL2_LIBRARY_PATH=/mingw64/bin/ -DPCB=X10 -DPCBREV=TX16S -DTRANSLATIONS=EN -DDEBUG=YES -DCMAKE_BUILD_TYPE=Debug ../
#
make configure
make -C arm-none-eabi -j`nproc` firmware
```

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### OS Version

Windows 11

philmoz, 2023-05-03:
What GCC version are you using?
This seems to be an issue in GCC 12 - https://github.com/raspberrypi/pico-sdk/issues/1029

mha1, 2023-05-04:
It's the one the scripts install.

GCC version:
```
$ gcc --version
gcc.exe (Rev11, Built by MSYS2 project) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Binutils version:
```
$ ar --version
GNU ar (GNU Binutils) 2.40
Copyright (C) 2023 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
```

The MSYS2 build environment on my older machine was also installed using the Wiki scripts but some months ago. It has a slightly older version of GCC (12.1 Rev. 11) and binutils (2.38) and doesn't show the RWX warning.

pfeerick, 2023-05-05:
The `LOAD segment with RWX permissions` warning was introduced with Binutils 2.39. https://www.redhat.com/en/blog/linkers-warnings-about-executable-stacks-and-segments

Unfortunately, MSYS always installs the latest packages, not pinned versions, so the environment build script that works today will not necessarily work tomorrow, and if you install it at a latter time on another system, you are pretty much guaranteed to get builds with minor variances because some package will be different :-/

mha1, 2023-05-05:
Yes, correct, but does this warning point to a real problem?

stronnag, 2023-05-05:
> Yes, correct, but does this warning point to a real problem?

No, it's generic to much embedded firmware with Gcc 12; the same happens if you build INAV with gcc 12 or later.

pfeerick, 2023-05-07:
Like any other compiler "waning", it's telling you there is a _potential_ misconfiguration issue, not necessarily an actual one. In this case, binutils is now warning that code and data are not separated into different segments. Is this a problem for us? Probably not...  it's pretty unlikely someone is going to craft some exploit that somehow allows code to be executed on the ETX platform... they've got the source code and DFU... can do what they want anyway 🤣 Should it be fixed? Maybe, but it can also make it so few more bytes of flash are wasted doing that partitioning. 

Should the build setup warning be fixed? given it's MSYS and is referencing some change in how Python is behaving, it's probably a precursor to something going spectacularly wrong if the warning isn't listened to, so probably yes. 

Did you take note of which step that was actually triggering that warning? Was it 
```
Fetching and installing Python package pyppmd
```
??

stronnag, 2023-05-07:
Simple to remove the link warning if it bothers anyone
```
    target_link_options(${target} PRIVATE "LINKER:--no-warn-rwx-segments")
```
For some `{target} `.

mha1, 2023-05-13:
Is any action resulting from this? I'm happy to close the issue if everything is hunky-dory

closed 2024-10-13 by mha1