EdgeTX/edgetx #3560: EL18 ，After replacing the SWH switch with a 3-stage switch, in the special function settings, it is impossible to select the middle state of the switch.

llwingok1, 2023-05-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

SWH, the original factory switch is a 2-stage switch, I replaced it with a 3-stage switch, the output is normal and it can output values correctly. However, when setting special functions, if it is associated with the 3-stage switch, the display of special functions becomes abnormal. Could an expert please help take a look?

### Expected Behavior

Fix this issue

### Steps To Reproduce

1、Change the SWH in the system to a 3-stage switch.
2、In the monitoring, test whether all three stages of the switch are working properly.
3、In the special function settings, select the SWH switch and set three special function entries for the three states of the switch.
4、When the switch is operated, it cannot correspond to the set special function entries.

### Version

2.8.2

### Transmitter

Flysky EL18

pfeerick, 2023-05-04:
This sounds somewhat similar to #3101

philmoz, 2023-05-04:
What does the keys debug page show for SWH (Radio Settings, Hardware tab, Debug section).

Does it show the correct up, middle and down icons like the other three pos switches?

llwingok1, 2023-05-05:
> SWH的按键调试页面显示什么（无线电气设置、硬件选卡、调试部分）。
> 
> 它是否像其他三个位置一样显示正确的上、中、下图标？

swh可以显示上、中、下，而且该通道对应的输出值也是正常的，能正常输出-100、0、100这三个值。我用这个通道，在bf固件里映射一些功能时，也是正常的，比如自稳模式、半自稳模式、手动模式，可以正常切换。
但是这个拨杆在edgetx里，设置特殊功能时，无法正常选中“中”的状态，当开关到“中”时，显示的是上，所以对应的逻辑条目会被选中，比如用这个摇杆控制声音。

llwingok1, 2023-05-05:
> 这个听起有点像#3101

我在系统中，已经将这个开关已经改成3POS了，但是还是不行。
[image]
[image]

pfeerick, 2023-05-05:
Please try going to Radio Settings -> Hardware -> Debug / Keys (at the very bottom). Does SWH show an up arrow when up, a dash when in the middle position, and a down arrow when down? i.e. Does it behave like SWB?

llwingok1, 2023-05-05:
> 请尝试转到 Radio Settings -> Hardware -> Debug / Keys（在最底部）。SWH 是否在向上时显示向上箭头，在中间位置时显示破折号，在向下时显示向下箭头？即它的行为是否像 SWB？

yes,

> Please try going to Radio Settings -> Hardware -> Debug / Keys (at the very bottom). Does SWH show an up arrow when up, a dash when in the middle position, and a down arrow when down? i.e. Does it behave like SWB?

> Please try going to Radio Settings -> Hardware -> Debug / Keys (at the very bottom). Does SWH show an up arrow when up, a dash when in the middle position, and a down arrow when down? i.e. Does it behave like SWB?

yes,it like swb。SWH show an up arrow when up, a dash when in the middle position, and a down arrow when down

https://user-images.githubusercontent.com/56796808/236414846-4246c285-6cff-40d1-9e7c-7e28ba6d1dce.mp4

pfeerick, 2023-05-05:
Thanks for that - that should narrow down where the fault in the code could be.

richardclli, 2023-05-06:
你用其他3段開關可以做到嗎?  是不是只有改了的3段有問題?
Did you have problem using other 3 pos switch?  Or only the modded switch not working?

llwingok1, 2023-05-06:
> 你用其他3段打开可以做到吗？是不是只有改了的3段有问题？ 您在使用其他 3 位开关时遇到问题吗？或者只有修改后的开关不工作？

原装的哪个是三段的？我已经不记得我换过几个开关了，我回去试试

richardclli, 2023-05-06:
> > 你用其他3段打开可以做到吗？是不是只有改了的3段有问题？ 您在使用其他 3 位开关时遇到问题吗？或者只有修改后的开关不工作？
> 
> 我查了下，EL18原厂的SWB、SWF、SWG是3段开关，我一会去试试看

SWF是3段的，我試不到你說的問題

pfeerick, 2023-05-06:
On both of my EL18, it still has SWB, SWF and SWG as three pos like the NV14, and I can use all three to trigger SFs in all three positions. 

However, in the production models, it appears SWC changed to 3 pos (from a momentary) and SWE changed to 2pos (from a momentary). I'm not sure how SWH would actually work as a 3pos - didn't know that sort of mod was supported.

llwingok1, 2023-05-06:
> > > 你用其他3段打开可以做到吗？是不是只有改了的3段有问题？ 您在使用其他 3 位开关时遇到问题吗？或者只有修改后的开关不工作？
> > 
> > 
> > 我查了下，EL18原厂的SWB、SWF、SWG是3段开关，我一会去试试看
> 
> SWF是3段的，我試不到你說的問題

我刚才做了个全面测试，发现了具体故障问题点：
1、故障复现条件：在特殊功能中，新建后，动作选择“播放声音”模式，工作异常，无法选择开关的中间状态“-”。选择“教练”模式，工作正常，可以正常选择到上、中、下三个状态。其余模式没有测试
2、故障涉及开关：SWE、SWH（原厂的3段开关SWB、SWF、SWG均正常，无问题）
3、故障复现方法：在复现故障的时候，不需要将SWE、SWH的物理开关更换为3段开关，可以使用原厂的2段开关模拟3段开关。只需要在系统中，改变需要测试的开关为3段，拨动的时候，在开关拨动的行程中间，用手扶住，强制停留，就可以测出中间状态“-”。

llwingok1, 2023-05-06:
> On both of my EL18, it still has SWB, SWF and SWG as three pos like the NV14, and I can use all three to trigger SFs in all three positions.
>
> However, in the production models, it appears SWC [...]

感谢，我也测试了，原厂的3段开关SWB、SWF、SWG均正常，无问题。出问题的是SWE和SWH，具体内容见上述回复

pfeerick, 2023-05-06:
Thanks for that tip, yeah, I can reproduce that with SWC configured as a 3POS

llwingok1, 2023-05-06:
> Thanks for that tip, yeah, I can reproduce that with SWC configured as a 3POS

EL18正面的部分SWA~D，我没有测试2POS——SWA、SWC、SWD，我只测试了正面的3POS——SWB，SWB是没问题的
EL18背面的开关SWE~H，这4个开关我都测试了，出问题的只有SWH和SWE
再次感谢大佬，希望能尽快修复该问题

richardclli, 2023-05-08:
Interesting that not all the switches has problem.  The code should be common.............  Maybe I am wrong.

richardclli, 2023-05-08:
@pfeerick I think I found the problem, it seems the switch definitions are all hardcoded.

https://github.com/EdgeTX/edgetx/blob/f800447cd5299dfec62f5e2c459ad49f762e15b7/radio/src/switches.cpp#L216-L237

pfeerick, 2023-05-08:
Oh, that again... that puzzled me when I first saw it... since you can change some switches between 2 and 3 pos... 🤪

richardclli, 2023-05-10:
> > Thanks for that tip, yeah, I can reproduce that with SWC configured as a 3POS
> 
> EL18正面的部分SWA-D，我没有测试2POS——SWA、SWC、SWD，我只测试了正面的3POS——SWB，SWB是没问题的 EL18背面的开关SWE-H，这4个开关我都测试了，出问题的只有SWH和SWE 再次感谢大佬，希望能尽快修复该问题

@llwingok1 你有在我的QQ群嗎?  我可以發你測試固件

richardclli, 2023-05-11:
I have tested, my PR works now.

llwingok1, 2023-05-15:
> > > 谢谢你的提示，是的，我可以用配置为 3POS 的 SWC 重现它
> > 
> > 
> > EL18正面的部分分SWA-D，我没有测2POS——SWA、SWC、SWD，我只测了正面的3POS——SWB，SWB是没有问题的EL18背面的开SWE-H，这4个开关我都测了，出问题的只有SWH和SWE 再次感谢大佬，希望能尽快修复该问题
> 
> @llwingok1你在我的QQ群里吗？我可以发你测试固定件

Edgetx开源控交流群。我在这个QQ群里，不知道大佬说的是这个吗？

richardclli, 2023-06-09:
> > > > 谢谢你的提示，是的，我可以用配置为 3POS 的 SWC 重现它
> > > 
> > > 
> > > EL18正面的部分分SWA-D，我没有测2POS——SWA、SWC、SWD，我只测了正面的3POS——SWB，SWB是没有问题的EL18背面的开SWE-H，这4个开关我都测了，出问题的只有SWH和SWE 再次感谢大佬，希望能尽快修复该问题
> > 
> > 
> > @llwingok1你在我的QQ群里吗？我可以发你测试固定件
> 
> Edgetx开源控交流群。我在这个QQ群里，不知道大佬说的是这个吗？

是的，我是IQ博士

PR #3568 (2023-06-22): fix: 3POS switches on certain radios not detecting all switch positions

Fixes #3560

The ultimate fix is https://github.com/EdgeTX/edgetx/pull/2631 in milestone 2.10.0
closed 2023-07-03 by pfeerick