EdgeTX/edgetx #3565: Additional switch SI beeps on activation despite "NoKey" selection in Radio Settings-Sounds

MirlynG, 2023-05-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have enabled switches SI and SJ and connected two momentary switches. 
As expected, SI beeps when "All" is selected in Radio Settings-Sounds. However, it also beeps when "NoKey" is selected, which of course it should not.
Conversely, SJ does not beep when either "NoKey" or"All" are selected. 
Neither keys beep when "Alarrm" or "Quiet" are selected, which is of course fine.

### Expected Behavior

When Radio Settings-Sounds-All is selected I would expect both switches to beep when activated
When Radio Settings-Sounds-NoKey is selected I would expect both switches to remain silent when activated.

### Steps To Reproduce

SI pushbutton pressed - beeps when "NoKey is selected and also "All"
SJ pussbutton pressed - no beep in either of the above selections

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Using EdgeTX 2.8.3

philmoz, 2023-05-07:
I don't think there are any sounds played for switches within the core system.

Are you using a special or global function to play the sound for the SI switch?

MirlynG, 2023-05-08:
I think there are but no, not using any GF's or SF's

philmoz, 2023-05-08:
Do you have a folder in your SOUNDS directory with your model name (e.g. SOUNDS/en/modelname) and in that folder there are files named SI-up.wav, SI-mid.wav and SI-down.wav?

MirlynG, 2023-05-09:
Having looked at this again, it seems that the Inst. Trim Global Function comes with a beep automatically! I have set it up on the other two momentary switches (SH and SJ) and on a couple of other switches on various positions. Every one of them beeps. Can this be chaged so that Inst.Trim is  covered by the NoKey option in Sound settings? Thanks.

pfeerick, 2023-05-09:
That probably isn't a good idea. By definition, NoKey means "No Key beeps", not "beeps because you've activated a SF/function. It's like asking for the Play Sound function to be also quietened when NoKey is selected instead when in "Alarm" or "Quiet". It would be better to have an option for Instant Trim to play a sound or not - it doesn't have any other options, so that _might_ be possible to do.

MirlynG, 2023-05-09:
Hadn't thought of it that way but your suggestion would be perfect if possible.  Thanks