EdgeTX/edgetx #3569: iNav widget is not working on Boxer

kirek007, 2023-05-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Using the latest version of https://github.com/iNavFlight/OpenTX-Telemetry-Widget caches the widget beacause of missing table API (according to https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/86)

### Expected Behavior

Widget should work.

### Steps To Reproduce

1. Install widget
2. Start widget 
3. It crashes if there is at least one LOG file
4. It might not crash on startup with latest version, then it will crash at log playback function because files are not listed due to table API error.

### Version

2.8.0

### Transmitter

Radiomaster Boxer

### Operating System (OS)

Windows

### OS Version

11

3djc, 2023-05-09:
Widgets author clearly indicates Boxer is not supported: https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/86#issuecomment-1511884625

kirek007, 2023-05-09:
> Widgets author clearly indicates Boxer is not supported: https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/86#issuecomment-1511884625

It would be working just fine, but table api is not included for bw radios according to the Lua references.

pfeerick, 2023-05-09:
I don't think that is the issue. table has *never* been included for B&W even on OTX, and it has worked on other B&W radios. :shrug:

kirek007, 2023-05-09:
> I don't think that is the issue. table has _never_ been included for B&W even on OTX, and it has worked on other B&W radios. 🤷

So, for now, simply remove` table.sort()` function made the widget work again. I'll look at changes in this function. Maybe they were using different functions to sort by date then

prelator, 2023-05-13:
I tested the latest version of the telemetry widget on my Boxer and it appears to work now even when logs are present. Hopefully this continues to work with no more errors.

closed 2024-04-08 by pfeerick