EdgeTX/edgetx #3572: more than 32 Channels with external Module

r41065, 2023-05-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Using FrSky Transmitters and a second Transmit Module to get more Channels. 

With D16 Protokoll there are 16 Channels to internal Modul and Channel 17..32 are possible to external Modul.

With Access you can set 24 Channels to internal but external is limited to 17..32 also.

### Describe the solution you'd like

is it possible to Start channels for external at a higher Number that you can get Channel 25..40 ( maybe 48 if there will be an external ACCESS modul in Future ) to work?

### Additional context

(   Testet with X9Lite because its the only Access Transmitter of my own till now )
[image]

3djc, 2023-05-09:
Memory wise, it is clearly impossible on BW radio without giving somethign else away

r41065, 2023-05-10:
in Companion its also not working for X10Express, also limited to 32 Channels for 2 different Transmit Moduls.

( I build model ships with many functions, but without complicated mixers / only the controls per flight phases assigned to the channels... ) 
Just testing which options are possible in the future.