EdgeTX/edgetx #3579: Inconsistent use of  RTC_BACKUP

3djc, 2023-05-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Sorry, it is going to be a little long, but I think some history is needed here.

Historicaly, otx radio were using on board eeprom for model storage. In case a critical software or hardware error happens (ESD,..), the watchdog reset the radio (emergency rmode) and model data was read out of eeprom. That was deemed a safe and reliable way to store such data, even if that happened during a flight.

When Horus came out, it was the first otx radio to use sdcard for model storage.  Given the number of issues existing with SD, it was deemed unsafe to rely on sdcard for critical model data in case emergency mode, so the RTC_BACKUP was introduced to safeguard critical model data in a special section of STM32F4 memory, so that in the event of emergency mode, data needed to safely keep your model flying despite the reset would be safe and not depending on sdcard.

Comes EdgeTX and yaml storage, and now all radio use sdcard for model storage, but only colorlcd radio have RTC_BACKUP set, when F4 based B&W radio really should have it too (F2 does not offer enough storage in the backup area to be workable)

### Expected Behavior

F4 bw radio do have RTC_BACKUP too

### Steps To Reproduce

Not applicable

### Version

Other (Please specify below)

### Transmitter

Other (Please specify below)

gagarinlg, 2025-01-24:
@raphaelcoeffic didn't you fix this?

pfeerick, 2025-01-24:
Might actually be @3djc, but wasn't tagged.

pfeerick, 2025-01-24:
#5686

gagarinlg, 2025-01-24:
So, unfixable for F2 AND F2 support will be removed with 3.0 anyways. I am closing this.

closed 2025-01-24 by gagarinlg