EdgeTX/edgetx #3583: V 2.8.3 Flysky FS-CAT01 altitude sensor

1RicMcK, 2023-05-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I updatged from 2.8.1 to 2.8.3 and the FlySky Cat01 altitude sensors do not read correctly.  Initally they say + or - 2-3K meters. Will offset if units are 0-.  Data reads max altitude of over 20Km during any given flight and does not give any intermidate readings.   2.8.1 works fine.

### Expected Behavior

Should be able to offset ot "zero" and should transmit realtime data 0-1,000Ft, and properly record max altitude.  Works fine with V2.8.1

### Steps To Reproduce

4

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### OS Version

2.8.3

richardclli, 2023-05-12:
The calculation method of alt from pressure is changed by Flysky to align with their own implementation.

https://github.com/EdgeTX/edgetx/blob/3692718e8ad383c6703a8b7f14d47bf645b3896b/radio/src/telemetry/flysky_ibus.cpp#L559-L617

And they have tested and verified the change.

richardclli, 2023-05-12:
@Xy2019 Do you have any comments?

1RicMcK, 2023-05-12:
Thank you. No, I don't have any comments as I'm not a programmer. I just
wanted to report a bug I found. I will wait until 2.9 comes out to update
my FW.

Thank You
Rick McKenney
"Change the dream and our actions will follow it"  Lynne Twist

John-4565, 2023-05-14:
I also noticed the same weird altitude reporting from my Flysky FS-CAT01 when upgrading EdgeTX from 2.8.2 to 2.8.3. The upgrade also messed up a number of other sensors such as Tmp2 and Pres. I downgraded my Radiomaster TX16S back to EdgeTX 2.8.2 and all was good again.

1RicMcK, 2023-05-14:
John, EtAl. I also down graded to 2.8.1,but have not tested it again. I
assume that everything will be OK.

Rick McKenney (via phone)

richardclli, 2023-05-17:
Please specify the chain of equipments that you use.

TX16S -> MPM? -> What receiver -> CAT01

John-4565, 2023-05-17:
> Please specify the chain of equipments that you use.
> 
> TX16S -> MPM? -> What receiver -> CAT01

The equipment I am using is a Radiomaster TX16S -> Internal MPM on F/W 1.3.3.20 TAER -> Rx is a FlySky FS-iA6B -> Altitude sensor is a FlySky FS-CAT01. Let me know if you need more detail.

1RicMcK, 2023-05-17:
I'm Running Radiomaster TX16S 4 in 1. \ mm-stm-serial-aetr-v1.3.3.20. \ RX FlySky FS-iA6B. \ Altitude sensor is a FlySky FS-CAT01.

CharlesGauci, 2023-05-18:
Even I had to downgrade my Flysky NV14 to 2.8.2 because of the FS-CAT01 issue. I also decided to shelf my Flysky EL18 for the second time because neither the temperature, nor the voltage and not even the altitude sensors are working properly. 
I mentioned this problem in PR number #3467 which is related to PR number #3386.

stefanpeter218, 2023-06-04:
I've observed the same behaviour with latest edgetx and the cat01 altitude Sensor. Verified the bug with different receivers (FS-iA10b, FS-iA6b) and transmitters (Radiomaster TX16S and Zorro).
This is very annoying and makes me thinking of switching back to opentx. 
With the previous release of edgetx I had the issue that global variables couldn't be changed, what caused me to update. Now this mess.

gagarinlg, 2023-06-04:
As was stated before, that change came directly from Flysky. If it does not work here, for whatever reason, we have to roll back that change.

stefanpeter218, 2023-06-04:
Dear gagarinlg, this was my thought. The calculation, e.g. in 2.7, seems to differ much from the 2.8.4 version (I couldn't find the 2.8.3 version .cpp).
Maybe flysky confused with afhds2a and afhds3 sensors?

gagarinlg, 2023-06-04:
I have no idea, I did not went into the rabbit hole of EdgeTX telemetry, yet.

richardclli, 2023-06-05:
> Dear gagarinlg, this was my thought. The calculation, e.g. in 2.7, seems to differ much from the 2.8.4 version (I couldn't find the 2.8.3 version .cpp). Maybe flysky confused with afhds2a and afhds3 sensors?

I think The current Flysky implementation aligns the reading of this sensor with their stock radio with their stock firmware.  As for your information, alt. is not detectable directly by sensors, it is calculated using pressure.  Flysky stock firmware uses the same table lookup to do the alt. transformation and the definition is the alt above sea level.

For Flysky sensors, there are quite a lot of historical problems in the code.  Third-party sensors, and flight control make use of the same ibus telemetry.  When Flysky try to align the representation between their own products and EdgeTX, they discovered there a list of sensors that is never produced by them.  I asked them not to touch that part and only fix their sensors and align with their own products.

I have been following these for months already but still cannot managed to see how to get the best out of both sides.  We may need a lot of help and time to get these resolved properly.

RNIR, 2023-06-09:
Just to confirm, with 2.8.3 I also found altitude readings for flysky unusable (garbage values, not just slightly different than before).
Already ordered new Sensor, because I thought it got damaged, but now found the same behaviour with a different model.
I am using TX16s, 4in1 internal module, FSiA6b RX, FS-CAT01 sensor.

richardclli, 2023-06-13:
I just verified for the FS-CAT01 sensor connected to AFHDS3 receivers, the reading are the same from PL18 with stock firmware and EL18 with EdgeTX.  Will try to verify the reading from AFHDS2 receivers via NV14 and MPM afterwards.

PR #3582 (2023-08-04): fix: Some AFHDS2A/AFHDS3 GPS/ALT telemetry handling issues
Fixes (partially) #3543 
Fixes #3583
closed 2023-08-04 by pfeerick