EdgeTX/edgetx #3588: Voice track does not play

melvinchow, 2023-05-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

playing the voice track in file navigator, or triggering via switch, results in silence.  Other voice tracks are working fine, other than the one uploaded to this bug report.

### Expected Behavior

hear voice when playing "timres.wav" sound track via file explorer or triggered by switch

### Steps To Reproduce

SYS Button > SD-HC Card > SOUNDS > Long Press "timres.wav" > Play

OR

Assign "timres.wav" to a Special function switch > Toggle switch to play track

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### OS Version

2.8.3

### Anything else?

[timres.zip](https://github.com/EdgeTX/edgetx/files/11459390/timres.zip)

pfeerick, 2023-05-12:
Duplicate of #3134

But I understand why this was missed as it basically doesn't come up when you search for Amber - which is where the timres.wav is from. Basically, as mentioned in https://github.com/EdgeTX/edgetx/issues/3134#issuecomment-1417218199 there are three files in the Amber pack that for some weird reason are uLaw encoded, and thus will no longer play since support for aLaw and uLaw encoded wav files was removed. 

[image]

Let me know if these work for you...
[fixed-amber-sounds.zip](https://github.com/EdgeTX/edgetx/files/11460286/fixed-amber-sounds.zip)

melvinchow, 2023-05-12:
Fixed sounds are working, thanks.

closed 2023-05-12 by melvinchow