EdgeTX/edgetx #3593: Companion update function does not work in Linux Mint

icebreaker-ch, 2023-05-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I'm using Linux Mint Linux Mint 21.1 Cinnamon and EdgeTx Companion 2.8.3
Using the "check for updates..." function results in an error box:

Unable to download Meta-Data.
Error:6
SSL handshake failed.

 Trying to update results in several errors and the update fails
.
[image]
[image]

### Expected Behavior

Companion should search for updates and offer to update the out-dated versions

### Steps To Reproduce

1. Start EdxgeTX Companion
2. Press "Help" -> "Check for updates..."
3. Press "Ok"

### Version

Other (Please specify below)

### Transmitter

FrSky X12

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.1 Cinnamon

mha1, 2023-05-14:
@elecpower Hi Neil, I think this is an OpenSSL version mismatch. User reports 
```
$ openssl version
OpenSSL 3.0.2 15 Mar 2022 (Library: OpenSSL 3.0.2 15 Mar 2022)  
```

Isn't OpenSSL 1.1.1 lib bundled with the Companion installer?

pfeerick, 2023-05-14:
Unfortunately, if that is the case, it won't show up that way - doing a `openssl version` at the terminal tells you what version is installed/provided by the OS, not the version that's in the appimage bundle (and hopefully used by Companion - i.e. hopefully it's not using the OS provided version instead of the bundled version).

mha1, 2023-05-14:
How can one verify if it's bundled and installed correctly?

gagarinlg, 2023-05-14:
running
```
ldd <command>
```
and check which libs are being loaded

icebreaker-ch, 2023-05-14:
Unfotunately

`ldd EdgeTx_Companion_2.8.3-x86_64.AppImage `

does not work:

```
$ ldd EdgeTx_Companion_2.8.3-x86_64.AppImage 
	not a dynamic executable
```

pfeerick, 2023-05-14:
You may find out you need to mount the appimage, so you can inspect what is actually bunded with it... I'm not sure you can use `ldd` normally with it since it's a mounted filesystem bundle. 

i.e. 
https://askubuntu.com/questions/1231597/how-can-i-examine-the-files-inside-an-appimage

icebreaker-ch, 2023-05-14:
I extracted the AppImage. Seems libopenssl is not bundled within. Result of ls -R of the squashfs root in the attached file.

[AppImage.content.txt](https://github.com/EdgeTX/edgetx/files/11471473/AppImage.content.txt)

mha1, 2023-05-14:
Another user reported 2.9 Companion working on Linux Mint (2.8 wasn't working). @icebreaker-ch confirmed openSSL as not being bundled with 2.8 Companion. If this is considered a major problem in 2.8 I suggest to implement https://github.com/EdgeTX/edgetx/pull/2935 in 2.8.4

pfeerick, 2023-05-14:
It'll have to be 2.8.5, but #2935 should indeed resolve that.

icebreaker-ch, 2023-05-16:
@pfeerick : Thanks for your work and the link. Your fix solves the problem for me. I also extracted the contents of the new AppImage and considered libssl.so.1.1 now is bundled with the AppImage. Everything OK!

pfeerick, 2023-05-16:
I just cherry-picked Neil's PR into 2.8. But I'm glad it's working for you now. I'll close this issue now then since it won't auto close. You're basically running Companion 2.8.4 + that fix, so it should be safe to use until 2.8.5 is released.

closed 2023-05-16 by pfeerick

wumfi, 2023-06-28:
> @icebreaker-ch Would you like to give https://github.com/EdgeTX/edgetx/suites/12890499788/artifacts/695626935 a try? It's the latest linux companion build for the 2.8 branch and will form part of v2.8.5, and includes the abovementioned fix.

Hi.  I appreciate that this won't be fixed until v2.8.5, but I'd also like to get this patch.  However, the link provided here doesn't work.  How would I go about getting this patched release?

mha1, 2023-06-28:
Those are development artifacts and are not meant to be patch versions. Hence they live a short life and are deleted after a few days of generation to avoid blowing up the servers. Please be patient and wait for the 2.8.5 release. Of course you can do a home brew version yourself based on the current 2.8 branch. You'll find more about setting up build environments in the EdgeTX Wiki at https://github.com/EdgeTX/edgetx/wiki.

icebreaker-ch, 2023-06-28:
@wumfi alternatively you could download the latest build of companion by navigating e.g. "Actions"->"Linux Companion" or "Actions"->"Windows Companion 64 bit" select latest workflow run "Artifacts"-> "edgetx-cpn-linux-merge",
illustrated here https://github.com/EdgeTX/edgetx/issues/3621#issuecomment-1560777883 by @pfeerick

wumfi, 2023-06-29:
@icebreaker-ch Thank you!