EdgeTX/edgetx #3597: PPM: first channel omitted in PPM stream

mha1, 2023-05-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The first channel is not output in PPM streams. E.g. PPM set to CH1-CH8 will output 7 channels (CH2-CH8).

### Expected Behavior

All channels are output in PPM stream

### Steps To Reproduce

PPM settings for external module:

[image]

PPM stream with internal RF set to OFF and external RF set to PPM with above settings - `NOK`
All sticks at neutral. Channel monitor confirms alls channels 1 to 8 at 1500us. Notice there is only 7 channels transmitted.
[image]

Channel 2 stick set to minimum, channel monitor confirms 988us. Notice The first channel in the PPM stream reflects channel 2. Channel 1 is missing in the PPM stream
[image]

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

nightly tx16s-d6e9472.bin

koaxheli, 2023-05-17:
Channel 2 stick and channel 8 stick set to minimum, channel monitor confirms 988us. Note: The first channel and channel 7 in the PPM stream correspond to channel 2 and channel 8. 
Channels in PPM stream are shifted.
[IMG_021.ZIP](https://github.com/EdgeTX/edgetx/files/11496569/IMG_021.ZIP)

mha1, 2023-05-17:
@koaxheli can you please test https://github.com/EdgeTX/edgetx/suites/12931660122/artifacts/698676182

koaxheli, 2023-05-17:
@mha1,
Have tested the firmware tx16s-a3b5000.bin everything OK.
But when I rebuild my new FW with Github
then the error with the shift at the External RF
with PPM is still present.

mha1, 2023-05-17:
Thanks for your fedback.

This is an issue, no solution. I offered a solution (PR https://github.com/EdgeTX/edgetx/pull/3598) which you tested ok as it seems. The PR is waiting to be merged. As long as it isn't  merged into main you'll still see the problem when building main. Just wait.

PR #3598 (2023-05-27): Fix: first channel omitted in PPM stream
fixes #3597

PPM settings for external module:

[image]

PPM stream with internal RF set to OFF and external RF set to PPM with above settings - `OK`
All sticks at neutral except CH1 set to Min. Channel monitor confirms all channel 1 at 988us, channels 2 to 8 at 1500us. Notice  all 8 channels correctly transmitted, timing as expected.

[image]

PPM OFF to PPM ON without glitches -  - `OK`
[image]
closed 2023-05-27 by pfeerick