EdgeTX/edgetx #3600: Improve splash screen customization for monochrome radios

ggoraa, 2023-05-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I don't quite like that I need to use the companion and fiddle around with it just to set a splash screen image, which will be overwritten the next time I update using Buddy.

### Describe the solution you'd like

A solution is to store the image on the SD card in the same way colorlcd radios do, so it gets loaded at runtime.

pfeerick, 2023-05-16:
I don't believe this will be possible - there is simply not enough memory on those radios to load the image at runtime. 

@raphaelcoeffic I think this is what you said when I was talking about this before?

3djc, 2023-05-16:
and also the timing of things, splash is dispalyed while radio is initialised and doesn't yet have access to sd

ggoraa, 2023-05-16:
Understood
Well, another way of doing can be through the bootloader. Like, you load the image file, and through the bootloader you "flash" it to flash storage, which I think improves the experience of flashing the splash screen image

pfeerick, 2023-05-17:
That is something we could possibly consider now, now that we have more than 600bytes of free flash memory for the smallest targets 😆

ggoraa, 2023-05-17:
lmao 😆
well, that's why cloud build is now getting developed

pfeerick, 2023-05-17:
That really only fixes part of the problem. And to be honest, I don't like being solely reliant on cloud build to fix any problems... isn't much use when the "cloud" is down, internet is down, don't have internet access, etc.

ggoraa, 2023-05-17:
we can always drop F2 radios 🤷

pfeerick, 2023-05-17:
You could always be made to disappear by a random group of F2 owners. That's like 1/3 of the B&W radios. :-P

pfeerick, 2024-01-08:
Changing the name / what requested made this a duplicate of #162 😆

closed 2024-01-08 by pfeerick