EdgeTX/edgetx #3604: Add bluetooth support for Radiomaster Boxer

JerryLutor, 2023-05-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The radiomaster boxer, like the MK2, has a bluetooth port. How problematic is it to add support to the firmware? building the firmware with the "-DBLUETOOTH=YES" parameter does not work

### Describe the solution you'd like

building the firmware with the "-DBLUETOOTH=YES" if support needed

3djc, 2023-05-17:
According to RM documentation there is no BT port on Boxer. Where does your belief there is come from ?

3djc, 2023-05-17:
What do you plan to do with it ? If needed I can probably do it, but in most cases, the serial port is better and more flexible

JerryLutor, 2023-05-18:
plan to use for bluetooth head tracker .
if it is possible to get rid of the extra wire, why not use this chance. this is my case. here the question is not only about me, if there is such an opportunity, others can also apply it to other cases.

3djc, 2023-05-18:
Ok, I'll try to explain why it matters:
you currently have access to that port, it is shown as AUXunder SYS/Hardware, that you can set to SBUS trainer for example
[image]. You can use standard BT chip to make  a transparent link.

What the BT option does is only allowing to use an FrSky BT module, which uses a specific protocol. Those BT modules are I think discontinued

So do you have an FrSky module that you need to use with Boxer ?

JerryLutor, 2023-05-18:
The module has been ordered and is on the way.

closed 2023-05-22 by pfeerick

robschii, 2024-01-02:
One question to this. I have a hc06 and hc05 bt module. both are configured but no telemtrie is shown in the app telemtry viewer. Boxer is on aux1 set up to telemtrie mirror and the BLE port is used.

i did not get any data.   anonther person from the facebook group has a working device. another friend has a tx16s and also a working device.

i need some help

pfeerick, 2024-01-02:
RX is connected to TX, and TX to RX? Any lights show on bluetooth module?

rotorman, 2024-01-03:
I would suggest to test first with a cable to see if the telemetry functions as expected, then change to wireless communication with HC05/HC06. If your device uses USB, then use e.g. 3.3V logic level FTDI cable, such as https://www.sparkfun.com/products/9717 or basically any other USB-to-serial adapter with 3.3V logic levels.

robschii, 2024-02-12:
> RX is connected to TX, and TX to RX? Any lights show on bluetooth module?

phone can connect to hc06, led is solid red when connect.  i connect via telemtry viever but nothing is shown. i check the wires .

nothing is shown in the app, all sesnors are working in telemetry screen.  aux1 is telemtry mirror. could it be the baudrate 115200?

Tuttle13, 2024-05-18:
@JerryLutor did you ever get this working? Hoping to do the same thing for head tracking, got the 05 board, but not sure what next in terms of software settings

CDubFPV, 2025-02-03:
Also checking @Tuttle13 @JerryLutor 
I have the Seeed Studio XIAO nRF52840 Sense and Arduino Nano 33 BLE Rev2 working in Headtracker Software
I also have the HiLetgo ESP-WROOM-32 ESP32 ESP-32S Development Board but scared to wire it up but looks like I will just have do to it and just leave it installed until I see a solution.