EdgeTX/edgetx #3609: Change trainer modes from Master / Slave to Trainer / Student

phileville, 2023-05-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The terms Master & Slave are not so descriptive and can be a bit confusing when configuring trainer mode. One could argue that we actually have it backwards as the Slave device is overriding the Master device. Trainer and Student is a clearer description for the modes.

### Describe the solution you'd like

Change the configuration settings from Master to Trainer and Slave to Student.

3djc, 2023-05-19:
The issue is some clubs have training planes bound to teachers radio, others where the students come with their model (and therefore their radio is bound to model). That's why this words where choosen

phileville, 2023-05-19:
But in that case where the student's radio is bound to the plane and configured as Master, with the instructor configured as slave, how does the the instructor take over when the student is in trouble? The SF to activate / deactivate trainer mode is on the Master radio.  Or is there a different way to configure this?

3djc, 2023-05-19:
I was using a Graupner when I was trained, and my trainer a Multiplex. The trainer switch was on my radio, more often than not, it was me requesting him to take over, but a few time he also asked me to give him back the plane. That works fine, the added benefit was that I knew my plane exactly when I graduated.

While I haven't tried it, you can probably use a slave channel to trigger trainer on master radio. Can try it if you want

ror1948, 2023-05-20:
my vote is leave as is...master/slave is standard for all Tx. Let's not just change for the sake of change.

Eldenroot, 2023-05-20:
Leave it, do not change

pfeerick, 2023-05-21:
I would probably have to agree with the "not change for the sake of change" sentiment. While I do agree that *some* other systems often refer to it as student and trainer radio, the convention for OTX and now ETX has always to refer to the primary radio as the master, and the secondary as the slave... as that makes more sense than primary and secondary ;) But yes, sometimes it is in reverse, the student's radio is actually the transmitting one, and the trainer's radio is the "slave", which would then introduce confusion. IMO we would have to gradually transition over in the documentation and ensure there was plenty of videos and guides, etc, using that wording before we could even contemplate changing it. And even then there just seems like there may be too much room for confusion?

ReneNwnhs, 2023-05-21:
As far as i can see the titles trainer/student are causing the confusion. Trainer ánd student are roles performed by persons handling a device. All that these actors do, is creating input that the device is interpretating based on rules. Only one device is in control of the model and only one device is mastering the communication. And the communication is what its all about. The device mastering, allowes incoming signals (from a tx acting as a rx or a 3,5 mm jack) to override its own input to the tx controling the model. Ór using these inputs (TR..) for other means or even not at all. Even if a student is handling the master device, this device is interpretating the rule that "If there is an incoming trainersignal, it has priority over the studentsignal". If there is no signal for whatever reason (communication-failure would be catastrofic), the student is in control. The device is still the master, no matter what.

phileville, 2023-05-21:
Thank you everyone for your thoughts and feedback on this issue. It gave me some additional perspectives that I had not considered.  Based on that I would agree that we should just keep it as it is. I will close this request.

closed 2023-05-21 by phileville

Schlaegel, 2023-05-23:
I've been bothered by the slavery terms in tech. The terms "primary" and "secondary," which were used by [pfeerick] to explain master and slave are general enough to be useful in many contexts, while also avoiding really horrible slavery analogies.

ReneNwnhs, 2023-05-23:
It's all about a relationship between devices in a communication scenario. Neather is learning or teaching the other anything, so ... What is this supposed to accomplish by changing clear terms into misty words?

ror1948, 2023-05-23:
oh but it is for buddy box setup and yes there is a teacher and student.
i say it is ok as is but if we must change.