EdgeTX/edgetx #3612: Max setting for bar style telemetry timer doesn't fit on 128x64 size screens

ulfhedlund, 2023-05-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

[image]

When setting the max time for a timer bar in the telemetry pages the hh:mm:ss field is partly outside the screen. It's a bit too far to the right to fit.

Tested on Xlite and X7 and seems to have been like this for quite a while, in OpenTX as well.

### Expected Behavior

The entire field should fit on the screen.

### Steps To Reproduce

1. Go to the Display screen
2. Set one of the screens to bar style
3. Choose to display one of the timers
4. The max. time field is only partly visible (the seconds are outside the screen)

### Version

2.8.2

### Transmitter

FrSky X-Lite / S / Pro

philmoz, 2023-05-20:
This has already been fixed. I expect it will be in 2.9.

pfeerick, 2023-05-21:
Yes, was fixed back in https://github.com/EdgeTX/edgetx/commit/80945c3874991229cb7b57bcfbe3586a0966c6b6

And yes, it is a bug from OTX. Shows how many people use bars and timer 😆

closed 2023-05-21 by pfeerick