EdgeTX/edgetx #3613: QX7 (XJT) - FrSky D8 loss of connection

nmw01223, 2023-05-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have an FrSky QX7 and a number of old whoops with SPI FrSky D8 receivers. Some time ago I updated the QX7 to the latest OpenTx and then found that on power up, they usually (not always) did not connect. The details are below, but I have just changed to latest EdgeTx (2.8.4) to see if the issue was solved there, but it is exactly the same.

With the radio powered, power the quad, and it doesn't connect - probably 8 times out of 10. On the other 2 times, it does. This only happens with the internal XJT module in use, and set to D8. Other quads on FrSky D16 using the internal module have no issue, and if I use an iRangeX 4in1 JR bay module instead, the quads connect every time. I do not think therefore it is a faulty XJT module as it works fine for D16. Its firmware has not being changed - the problem occurred after a change to the main Tx firmware.

The way out of it is quite simple - go into the main setup menu, press bind, wait a few seconds, press bind again (to stop it), and it connects fine. Once connected it stays connected, and note also there is no need to put the receiver into bind mode.

The quad can be powered down and up again, and it will reconnect fine, but if the QX7 model is changed, or if the transmitter is powered off and on again, the problem reoccurs.

I am pretty sure it coincided with the firmware update, never a problem before that.

I hoped it would be solved with a change to EdgeTx, but it isn't.

NOTE EdgeTx version is actually 2.8.4 not 2.8.2, but selector below does not include it.

### Expected Behavior

Power on radio, power up quad, and it should connect every time - assuming the quad has already been correctly bound.

### Steps To Reproduce

Power on radio, power up quad, and providing the protocol is D8 on the internal XJT module it probably will not connect. Briefly turn bind on and off again, and it will then connect. If the radio os powered off, or the model is changed, the problem will reoccur. If only the quad is powered off and on, it won't.

### Version

2.8.2

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

### Operating System (OS)

Windows

raphaelcoeffic, 2023-09-05:
@nmw01223 could you please re-test with the latest `2.9.0` release?

nmw01223, 2023-09-06:
OK, will test this within the next 2 days (currently away).

dynaMITE29, 2025-11-29:
Hey I am having the same problem, was this ever solved?

FrSky Qx7 (ACCST)
Mobula6 (the original with FrSky SPI) on D8

Upgraded from OpenTX to EdgeTX 2.11.0 (the final version compatible with the Qx7)
Whenever I power on the TX it forgets the bind and have to rebind as @nmw01223 described above.

I also upgraded my mobula6 to Betaflight to 4.3.2 which I believe to be about the last BF version to support my CRAZYBEEF4FR AIO and FrSky SPI.

As I understand it, where an AIO uses FrSky SPI the FrSky firmware has to be integrated into the BF firmware.
I have also read that the FrSky firmware on TX and FC/RX need to match.
Does this mean that we would potentially need to flash the Qx7 internal module firmware to match what is on the FC/ SPI RX?

And does that mean that if we downgrade the BF firmware to 4.2.x then we will potentially have the correct FrSky firmware integrated and it will all work correctly?

If this checks out we might be able to work out what FrSky firmware version is integrated into BF 4.3.2 (or whatever version you are using) then we could flash the corresponding firmware onto the Qx7 internal module?

Another complication is that all the FrSky internal module firmware versions are confusing as they do not have a specific Qx7 ACCST firmware download page as they do for the Qx7 ACCESS...
There is an XJT firmware download page which people on forums seem to reference as the correct firmware for the Qx7 ACCST, can anyone confirm this?

If anyone can clear any of this up it would be much appreciated!