EdgeTX/edgetx #3617: Wrong Telemetry Sensor Instance in exported model yml files

icebreaker-ch, 2023-05-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I took over my models from OpenTX configuration. The Telemetry-Sensor ID's and Instances are presented correctly in the Telemetry Page of the models (e.g. Id F103, Instance 27).
When I export a model to an yml file or write the configuration to the Transmitter, the Instances are different (Id 61699 (=F103 ok), Instance 250 (wrong)) and the Transmitter does not receive any Telemetry data. Looking at the model yml-file also shows the wrong instance number. After changing the Instance back to e.g. 27 or switching on the option "Ignore Instances", the Telemetry data is received by the Transmitter.

### Expected Behavior

Take over the instance numbers as they are and write it to the model yml files and to the Transmitter.

### Steps To Reproduce

1. Import a model with telemetry sensors from an OpenTX configuration file
2. Check the Sensor ID's and Instance numbers on the Telemetry page
3. Export the model to an yml-file or to a Transmitter
4. The Sensor ID is correct but the Instance number has changed, Telemetry does not work

### Version

2.8.4

### Transmitter

FrSky X12

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.1 Cinnamon

pfeerick, 2023-05-22:
This may have been fixed for 2.9 - would be a good idea to check a nightly firmware and companion to see if it is still present or if https://github.com/EdgeTX/edgetx/pull/3401 fixed this.

icebreaker-ch, 2023-05-23:
Ok, would like to check that. But: How can I get a nightly companion or a 2.9 version of the companion without self compiling it?
And why is the companion not in the Assets of the nightly builds?

pfeerick, 2023-05-23:
Because it's not built nightly - but when there are changes.

Companion (main branch): 
- Windows =>  https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain
- Mac => https://github.com/EdgeTX/edgetx/actions/workflows/macosx_cpn.yml?query=branch%3Amain
- Linux => https://github.com/EdgeTX/edgetx/actions/workflows/linux_cpn.yml?query=branch%3Amain

icebreaker-ch, 2023-05-24:
I confirm, the issue  is fixed in companion 2.9

closed 2023-05-24 by icebreaker-ch

pfeerick, 2023-05-25:
Thank you very much for persevering and checking on this... very much appreciated! 😍