EdgeTX/edgetx #3621: Same artifacts for nightly builds as for releases

icebreaker-ch, 2023-05-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When reporting issues, you are often asked to check, if the problem is already solved in the main/latest release.
While the transmitter firmware is available in the Assets of the nightly releases, this is not true for the companion software.
In my opinion, it is not reasonable for the average user, to compile the software by him/herself to have the latest version of the companion software.

### Describe the solution you'd like

The nightly builds shall provide the same artifacts as the official releases do, in particular the companion (edgetx-cpn*) should also be part of the nightly builds.

### Describe alternatives you've considered

1. Compiling the software by myself using gidpod (not THAT hard, but nothing for the avearage user)
2. Compiling the software by myself in  virtual machine (even harder)
3. Finding pre-compiled versions on the internet (not really feasable in my opinion)

pfeerick, 2023-05-24:
At present, this is not going to happen. There are builds for companion whenever there are companion changes, available here, no need to compile. They are not built on a nightly basis as 1) there are less changes to companion than there are to the radio firmware and 2) we are already using a LOT of hours of build time to do the builds we are already, without adding another ~4h every night for companion builds. 

Companion (main branch): 
- Windows =>  https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain
- Mac => https://github.com/EdgeTX/edgetx/actions/workflows/macosx_cpn.yml?query=branch%3Amain
- Linux => https://github.com/EdgeTX/edgetx/actions/workflows/linux_cpn.yml?query=branch%3Amain

icebreaker-ch, 2023-05-24:
Ok, I understand, the build strategy for the companion is based on file changes, not on a regular time base. But where are the artifacts for the companion of the last build? The link you provided, was of no help for me. I was not able to find the latest version of a companion there.The last change of the companion software happened yesterday.

pfeerick, 2023-05-24:
Sometimes a picture is better than words ;)

[image]

icebreaker-ch, 2023-05-24:
Thanks a lot for your help! Must have been blind not to see that... shame on me.

closed 2023-05-24 by icebreaker-ch