EdgeTX/edgetx #3623: Use Global Variables names everywhere in the GUI instead of certain places only

Neutral67fr, 2023-05-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I often have problem to set up a Special Function or Widget using a global Variable, because those functions don't show the names of the GV but their standard number (GFx) instead.
When we have many GV it becomes complicated to remember what GV is what...
In some settings the system use the GV names (such as Input, Mixes...) which is convenient to identify each GV.

### Describe the solution you'd like

I would simply like that the system uses the GV names when they are set, each time we have to refer to a GV, everywhere in the GUI.
Exactly as the source inputs (switches, sliders, pot.) when they have been renamed in the radio settings. 

### Additional context

[image]

pfeerick, 2023-05-25:
This will partially/mostly be implemented by #2219 when merged.

Fortesque0, 2023-06-28:
Hello,
Yes, it's necessary to see GV assigned name. For example to view value of GV using widgets, it's pretty crazy view on screen for example  GV1- GV2....instead assigned names.

PR #5526 (2024-09-11): feat(color): show global variable names in UI
Fixes #3623 

SF/GF edit:
[image]

SF list:
[image]

GV edit (title bar):
[image]

Source select (already in 2.11):
[image]
closed 2024-09-11 by pfeerick