EdgeTX/edgetx #3625: Jumper T16 switches interfer with each other.

jauniusbel, 2023-05-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

"Switches" window does not correspond to switches I'm turning.
SA as 3POS: up/down - correctly, middle - indicates "up".
SB as 3POS: middle position is indicated as "up" position. up/down possitions - correctly. When in middle position switch "SA" is affected when switching SB as follows: sa:up turns SB:up, SA:middle sets SB:middle, SA: down, does not change SB indication. If at the same time SA is switched to middle, then SB works correctly.
SC as 3POS: up,down - correctly, but middle indicated as "up" (if SD switched up).
SC as 3POS: up,middle,down - all correctly (if SD switched middle or down). But at the same time when SD is down, SC affects SD indications as follows:
SC:up -> SD:up, SC:middle -> SD:middle, SC:down -> SD:middle

SD as 3POS: correctly 

Switches test window responds as follows:
SA (same as "switches" screen)
SF: indicates "down", no reaction to switch.
If SE:down or SE:middle, SF works.
when SF up, SE does not change indications

### Expected Behavior

Indications matching actual button positions

### Steps To Reproduce

1. Open hardware/input/switches screen 
2. Move all the switches to up position.
3. Do actions as described.
4. open hardware/debug/keys screen
5. Move all the switches to up position.
6. perform actions as described.

### Version

2.8.4

### Transmitter

Jumper T16

### Operating System (OS)

macOS

### Anything else?

exact model: Jumper T16 Pro Hall
Had no problems with inputs in OpenTx

TramALot, 2023-05-25:
Test the switch with an ohm meter?

3djc, 2023-05-26:
Likely flat cable issue

jauniusbel, 2023-07-05:
After testing - indeed - this was a cable issue. Thank you!

closed 2023-07-05 by jauniusbel