EdgeTX/edgetx #3630: Option to ask before transmitting when powering up or switching models

factoryroll, 2023-05-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Radio transmitters should not be powered on without an antenna connected. Many transmitter modules have removable antennas. This presents the opportunity to damage a module accidentally.

### Describe the solution you'd like

I'd like to see an option that when a model is selected or the radio is first turned on, the radio prompts before beginning to transmit. It would be great if the prompt could also indicate which module (internal/external) is about to start transmitting. 

### Describe alternatives you've considered

1. Trying to remember to always have an antenna connected to both modules... not always successfully! 

### Additional context

If such a feature is introduced, there may need to some consideration for 'emergencies' where the radio is rebooted with a model in flight. Perhaps a solution would also be to introduce a configurable 'override' for this mode. Eg. If the aileron or elevator stick reaches maximum travel more than once in quick succession, the model begins transmitting on the previously selected module.

Eldenroot, 2023-05-29:
Good idea. Some brands have this.

pfeerick, 2023-05-29:
Sounds feasible... there's already facility for prompting for whether to use the internal or external antenna for some radios, at a global and model level basis, so this could probably be tacked onto that, or done in a similar manner. 

In as far as what would happen in emergencies/radio reboot - most likely the course of action would be to skip the check...

richardclli, 2023-05-29:
> Sounds feasible... there's already facility for prompting for whether to use the internal or external antenna for some radios, at a global and model level basis, so this could probably be tacked [...]

Just have a popup before powering on the module.

pfeerick, 2023-05-31:
Good point... there is no RF output while the model notes are shown as a pre-flight checklist (existing feature)... which will be more like a checklist in 2.10.

factoryroll, 2023-06-19:
Thnaks for the checklist suggestion. If possible, I think this issue would be best addressed as an optional system setting, rather than on a per-model basis.

55Laurent, 2023-11-21:
for info, Graupner TX ask for hf on or off at startup. If no option is choozen before 5s then the startup go ahead with hf on. This is helpfull when you just want to use the tx alone with no hf, for example for making some adjustment in the model program, this avoid to blow hf and drained battery power for no reason.

pfeerick, 2023-12-07:
As a thought (without having looked at the code for feasibility...)... is the module continuously turned on? Maybe it's *just* a matter of ignoring the request to turn the module on if in a state of no RF? Requiring a power cycle, model switch, and/or an option to enable (or disable, I guess) RF output in a menu? Personally I have no use for it, and would use a model checklist to remind the connect the antenna if needed, but if it can be made feasible, it would probably be a good option to have.

raphaelcoeffic, 2023-12-09:
> is the module continuously turned on? Maybe it's just a matter of ignoring the request to turn the module on if in a state of no RF?

In fact it is actively checked every cycle whether the settings corresponds to the current state. This means that we would have to actively prevent the module from activating (which is done to effect a reboot of the module). Also, quite some places in the code will use the module settings as configured in semantically equivalent way to "this module is running". So I predict that it would have a lot of unintended consequences.

pfeerick, 2023-12-09:
Fair enough. So it goes in the "too hard" basket for now since we'd be
actively fighting some of the core system design.