EdgeTX/edgetx #3645: Logical switches not saved

pierrotm777, 2023-06-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The logical switches from L30 to L37 are not saved.
L40 to 47 are saved

### Expected Behavior

When i shut down tmy Tx16s Mk2, these logical switches are not saved.
I use a compiled 2.9 firmware with the 1/6/2023 sources

### Steps To Reproduce

Never saved after power down

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows

pierrotm777, 2023-06-01:
I have found than the issue is when i use the TSwitch widget .
I use 2x8 buttons, with one witch use the L30 to L37 logicals switches and the second witch use the L40 to L47 logicals switchs.
When i open the first 8 buttons that erase the logical switches L30-37.
When i open the second 8 buttons that erase the logical switches L40-47.

pfeerick, 2024-03-05:
@pierrotm777 So are you saying there that the issue is actually a bug in https://github.com/Ziege-One/TSwitch or do you believe there to be an issue in EdgeTX still?

pierrotm777, 2024-05-06:
Sorry, I haven't seen your answer.
I think it's an edgetx issue.
I try to do some new tests this morning.

pierrotm777, 2024-05-06:
I test again to add Logical Switch and L30 to L37 aren't saved after handset power off.
As i understand i need to defined 8 Logical Switch like that:
[image]

Thanks for your help !
Pierre

pierrotm777, 2024-05-06:
[TSwitch.zip](https://github.com/EdgeTX/edgetx/files/15218624/TSwitch.zip)

pfeerick, 2024-05-06:
With what version of EdgeTX are you testing?

pierrotm777, 2024-05-06:
Ok, how can i configure and use this widget ?
Do i need to update the libgui.lua library ?

3djc, 2024-05-06:
Maybe you should see that with said widget author ?

pierrotm777, 2024-05-06:
I have followed this video from the author.
https://www.youtube.com/watch?v=PFdHdTHB5m0

pierrotm777, 2024-05-06:
I test the version 2.10.0 RC4

pfeerick, 2024-05-06:
I don't think that video is correct for the version of the script you are using... it seems it changed to using Sticky switches, and no, I don't thing you add anything yourself, the widget adds anything if it needs to. 

So what is the *actual* problem you have here?

pfeerick, 2024-05-06:
So do you want more than one button to control the channel position. i.e. three buttons could be used to mimic a three position switch? As it is not suited to that in it's current form... it is only good for single channel / momentary controls, as the second parameter is how long it should stay active for on it's own after you press it. i.e. 100ms, 500ms, etc. You would either want some more logic (although you are running out quick as it is consuming 8 logical switches with each instance of the widget) or a widget designed more for that purpose. 

I'll close the issue for now, as from what I can tell, it seems to be either an issue with the widget, or with how to use it. Please do let us know if you get anywhere with this, or a response from the author, as I know you have asked before at https://github.com/Ziege-One/TSwitch/issues/1 but didn't get any reply.

closed 2024-05-06 by pfeerick

pierrotm777, 2024-05-06:
Ok, Thanks you :-)
I need to work on this a bit.