EdgeTX/edgetx #3653: support for connecting a headtracker via bluetooth on radiomaster boxer

JerryLutor, 2023-06-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Thank you for the implementation of connecting the bluetooth module to the radiomaster boxer.

the module is flashed and configured according to the instructions

https://www.instructables.com/Bluetooth-on-Radiomaster-TX16S/

determined and work.

There is a problem with the connection, or I'm doing something wrong.

https://github.com/ysoldak/HeadTracker

Made:

In the menu system - hardware - bluetooth, the trainer is selected.
the address of the connected module was determined. The name for the module is specified as "---".
In the MDL menu - in the trainer section, master / bt is selected.
then did discovery. The poppy address of the connected head tracker was determined and the connection was made.
When you select the tracker channels in the model settings on the desired channels and go to channel monitoring, the change that comes from the tracker is not displayed.

### Describe the solution you'd like

Add debug data output when connecting the trainer, as it is done on TX16S.
since at the moment it is not clear this is a problem of the tracker or radio firmware.

3djc, 2023-06-03:
Do value change when you are connected in sys/trainer page ?

GarryBu, 2023-06-03:
And where can I get the boxer firmware that includes bluetooth?

JerryLutor, 2023-06-04:
> Do value change when you are connected in sys/trainer page ?

no, no value is changed.

JerryLutor, 2023-06-04:
> And where can I get the boxer firmware that includes bluetooth?

[firmware.zip](https://github.com/EdgeTX/edgetx/files/11644543/firmware.zip)
[image]

JerryLutor, 2023-06-04:
with the firmware of this bluetooth module, it will not work. close. sorry. I will assemble another head tracker.

closed 2023-06-04 by JerryLutor

3djc, 2023-06-04:
The Bluetooth option is for **FrSky** (or compatible) modules