EdgeTX/edgetx #3662: disable button on switch settings when you have a broken switch on your controller

jawz101, 2023-06-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I had a switch physically break on my TX16s which was my GPS Rescue switch.  Consequently I had to delete its programming so that it wouldn't intermittently trigger since the remaining bits of it would wiggle around in its housing.

### Describe the solution you'd like

I was wondering if there could be a way to disable switches w/in the setup screens so that it's easier to reenable the switch & retain its programming instead of deleting the switch's settings until the new part arrives.

### Describe alternatives you've considered

delete switch settings

pfeerick, 2023-06-06:
I've not checked how the code functions, but have you tried turning the switch off in the radio setup -> hardware page - i.e. change the type from 2POS/3POS to NONE?

philmoz, 2023-06-06:
> I've not checked how the code functions, but have you tried turning the switch off in the radio setup -> hardware page - i.e. change the type from 2POS/3POS to NONE?

You need to restart the radio after changing this setting.
Also the affected switch will still exist but always be set to 0 - so a switch check for SA- would still succeed (for example).

3djc, 2023-06-06:
I think thats probably an issue. To me, a disabled (set to NONE) switch should not be evaluated at all

gagarinlg, 2023-06-06:
Which is currently not an option. We need a switch state "invalid" which always returns false on all comparisons

3djc, 2023-06-11:
Since switch handling is basically completely rewritten in ADC refactor, this has been handled there. This comit: https://github.com/EdgeTX/edgetx/pull/2631/commits/c8e9437cd23349630de3c6fb4e5e4fd2c57fe177 adresses this issue.

- Special function are not trigered by 'none' switch
- In logical switches, 'none' switch is always false

PR #3715 (2023-06-25): switch-refactor-rb
Rebase of #2631 for merging, thanks to @gagarinlg 

This PR is likely to break a lot of things, so will need extensive testing and feedback. For a change, we're going to let this lose on the nightly users, early into the 2.10 cycle so it gets as much stress testing as possible, as well as to finally get it in ;)

This fixes #3662
closed 2023-06-25 by gagarinlg