EdgeTX/edgetx #3667: (feat) GUI enhancement, scroll up page title & top-bar together with the page

offer-shmuely, 2023-06-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have models with very complicated setup, 
it may take more than one screen per mixer
and may take more than one page for inputs
I find it very uncomfortable
we like to see more data on the existing page

### Describe the solution you'd like

I am suggesting a change that the title of the pages (i.e. Inputs/Mixers and others) 
as well of the top-bar will scroll up together with the page, and will not stay persistent

top-bar alternative: the top-bar will shrink upon scrolling

in this way, we get more space on the screen for the data we need, 
without add penalty to the touch operations

offer-shmuely, 2023-06-08:
Sample from safari on iphone
before scrolling 
The **_top-bar_** (url area) is **_big_**
The **_title_** (github + edgeTx) is **_visible and big_**
[image]

After scrolling up
The title  (github + edgeTx)  (github + edgeTx)
The **_top-bar_** (url area) **_shrieked_**

[image]

offer-shmuely, 2023-07-03:
Any opinion please.
Can it be implemented? Not a good idea?

offer-shmuely, 2023-07-26:
Hope it will be able to implemented sometime

offer-shmuely, 2024-09-06:
Still hoping...

offer-shmuely, 2024-12-12:
still hopping...

mhotar, 2024-12-16:
If I remember correctly you @offer-shmuely  mentioned the same problem I experience during scrolling through long pages. I described it already in discord so I paste it here as well:

If I can mention something here yet. I experience hard times when scrolling through list items with roller. These jumps from the last items on the page end to the back to the beginning are hard for my brain to accept 🙂 I am not against it in general but it lacks some visual aid for slow people like me. It would be nice, if possible, to add something like quick page scroll up/down instead simple jump. But maybe it is only me 😉

Your aproach could be a solution for my problem as well. If the page scrolls whole, including the page header, the first line is the one, which has allways the tittle bar above itself, so it can serve as a visual guide for jumps from the last row to the first one.

But. And there is always one. It will make the CLOSE button and tabs (in the page title) a bit harder to access without scrolling back when you are deeper in the page. Problem with the CLOSE button can be avoided simple with removing it completely (I am nor sure if it is planned to be there in 3.0). This could require to have "Main screen" group in 1st level and its subpages in 2nd Level present in planned Quick launch menu serving as an (Go home button"
Well, this are only my thoughts, I am not sure it fits to the planned concepts of the 3.0. I am sure @JimB40  will surprise us with something great.