EdgeTX/edgetx #3669: EL18: Audio output should be switch off when no sound is playing,

richardclli, 2023-06-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Audio is remain on all the time.

### Expected Behavior

Audio should switch off when nothing is playing.

### Steps To Reproduce

1. Install ELRS ext module
2. Use 2W output
3. Hissle sounds maybe heard in the speaker

### Version

2.8.4

### Transmitter

Flysky EL18, Flysky NV14

brainbubblersbest, 2023-06-09:
On x12s the interference is picked up over the headphone jack

pfeerick, 2023-06-09:
NV14/EL18 doesn't have a headphones jack, but yes, that would make it even worse / more noticeable. I can hear it on other radios like the X9D+2019 as it has a much louder speaker (thus *everything* is louder) and also the TX16S with ELRS gear if it's at higher power levels and the speaker is active, and have heard the odd report that it is worse for some than others, hinting at the audio shielding (or lack of) being part of the problem, just not sure why not with the EL18. Either way, the mute pin isn't being toggled for this radio family when the audio subsystem isn't in use, so that needs fixing.

brainbubblersbest, 2023-06-12:
> NV14/EL18 doesn't have a headphones jack, but yes, that would make it even worse / more noticeable. I can hear it on other radios like the X9D+2019 as it has a much louder speaker (thus _everything_ [...]

The mute pin is soldered/hardwired to grundsätzlich on x12s.
The Shutdown Pin on the other Hand is traced to the processor. Same for x10/s/express/t16.

I think this should be tried via Shutdown the Amp.
A Shutdown Amp cant reply the 900Mhz pickedup from anywhere and drive the connected speakers.

The humming Sound is existent on All Radios so far.
No matter if a crossfire or another 900mhz high TX Module is installed. 
Its like placing an old Cellular Device on a active speaker System while receiving an sms

PR #3670 (2023-06-15): fix(el18): Audio output should be OFF if no audio playing

Fixes #3669

Summary of changes:
closed 2023-06-15 by pfeerick