EdgeTX/edgetx #3673: Latest version 2.8.4

DonaldHillNZ, 2023-06-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Updating from 2.8.0 to 2.8.4 on a Radiomaster T16s causing the loss of return, page back, and other functions. Was trying to get Bluetooth loaded using the latest Lange version but it does exactly the same thing. Returning to 2.8.0 everything back to normal. Tried downloading and reloading the firmware. Same results

### Expected Behavior

Buttons to work

### Steps To Reproduce

Turn on the TX

### Version

2.8.4

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

10

3djc, 2023-06-10:
Looks like you flashed the wrong firmware for your radio/ What is the name of the file you flashed ?

DonaldHillNZ, 2023-06-10:
I'll have to get back to you in the morning. Sorry I've had covid recently and in bed. I did check the file and used t16 and also when I got the Bluetooth version I used the t16 as well.I'll recheck and let you know. I hope that's it but I have done this many times. But I guess we can do dumb things. Donald

rotorman, 2023-06-10:
@DonaldHillNZ T16 is for Jumper T16. If you have RadioMaster TX16S, you need tx16s firmware, NOT T16.

DonaldHillNZ, 2023-06-10:
Yep I know. I hope it is me being dumb. I'll certainly let you know in the morning. I was typing here t16 but I did know t16s. Can only blame dam covid if I've been that stupid.Donald

DonaldHillNZ, 2023-06-10:
Ok so I am dumb and have to apologise to everyone. Was the t16 and not t16s. Ive 2 x t16s's and 1 x t16 so you would think id know the difference. I do but all I can say in my defense is that I had a bad case of Covid recently and the bloody braion fog does exist. Sorry guys....

DonaldHillNZ, 2023-06-10:
as above

closed 2023-06-10 by DonaldHillNZ

pfeerick, 2023-06-11:
I know that feeling well! 😢 Rest up and I hope you get over it soon!

DonaldHillNZ, 2023-06-11:
Hi Peter,Thank you. Not at all pleasant. Looks like I might have long covid dam it. But thank you. Maybe more working on planes and things.Donald