EdgeTX/edgetx #3675: Companion Custom Screens allow 4 lines of "numbers" but Boxer only displays 3

plarkinjr, 2023-06-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In companion, you can set up to 8 items on a "numbers" custom screen (2 columns, 4 lines), but the Boxer only displays 3 lines (6 items).

### Describe the solution you'd like

Companion should not display, or offer to configure the 4th line of numbers (the last entry on both left and right columns).

### Describe alternatives you've considered

Leave the bottom 2 elements undefined.

philmoz, 2023-06-10:
The bottom row values are displayed at the bottom of the telemetry page in the inverted strip.
They will only display if there is telemetry data coming in - otherwise it display 'no data' in this section.

plarkinjr, 2023-06-19:
> The bottom row values are displayed at the bottom of the telemetry page in the inverted strip. They will only display if there is telemetry data coming in - otherwise it display 'no data' in this section.

Indeed it does show "no data" but it does not display the values programmed as the 4th line.    This screenshot shows me setting line 2 to Time and Batt, while line 4 is Batt and Time.  The radio simulation is accurate depiction of what the radio displays:   Line 2 (Time and Batt), while line 4 is inverted and displays flashing "No Data".  I would expect it to display Batt and Time as defined.   It does not matter what I set the 4th line to, it always shows "No Data". 

I'm also attaching a screenshot which shows that I have 3 lines configured, and there is only room for those, and the inverted line flashing "No Data".

[image]
[image]

3djc, 2023-06-19:
As said above, you need telemetry for the 4th line to be displayed, you must have active telemetry streaming, you do not, so no data is displayed instead. If you had' those would be displayed

3djc, 2023-06-19:
[image]

This is on a boxer WITH a receiver connected

pfeerick, 2023-06-20:
To put that into context on the simulator:

[image]

you can see as soon as I enable telemetry simulation, the bottom two lines appear - in this case it won't simulate fully as it's a ELRS model, but I think you get the idea. The bottom line is either the warning that there is no telemetry data, or the data configured the for 4th line. This may need to get clarified in the documentation, but it's not a bug or anything needing changing.

closed 2023-06-20 by pfeerick

plarkinjr, 2023-06-20:
Thanks for the feedback.  In fact I have no models which have telemetry sent from the model to the transmitter, so its out of context in my use-cases.      Yeah, at the very least, I guess clarification in the documentation (or limiting the choices for the last line to Rx streamed telemetry items only).