EdgeTX/edgetx #3678: Add Sbus In feature for External module

pierrotm777, 2023-06-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

NO, i should like just to use the Sport Pin for use it as SbuS Input pin, inverted or not .

### Describe the solution you'd like

NO, i should like just to use the Sport Pin for use it as SbuS Input pin, inverted or not .

3djc, 2023-06-12:
What radio ?

By the way, it is NOT possible on Taranis using Sport pin, but another pin (sport pin is not a standard io pin)

pierrotm777, 2023-06-12:
Hello 3djc, on a Radiomaster TX16S MkII .

pierrotm777, 2023-06-12:
If you can add this feature for transmitters witch accept, I should like that.
Thanks,
Pierre

Helipro, 2023-08-01:
> If you can add this feature for transmitters witch accept, I should like that. Thanks, Pierre

This feature works with a sbus receiver as a trainer and is under Radio Settings>Hardware> SBUS trainer

With aT16s you can also use a Bluetooth trainer
https://headtracker.gitbook.io/head-tracker/getting-started/wiring/option-2-wired/serial-bus-sbus/radio-master-tx16s-setup

closed 2024-02-26 by pfeerick